mne-tools/mne-python #3945: Error writing noise covariance

WilliamCollishaw, 2017-01-30:
code:
`noiseCov = compute_covariance(epochsList, tmin=None, tmax=0, method='shrunk',return_estimators=False, verbose=True, n_jobs=1,projs=projList)`
`mne.write_cov(subjectDirectory + URSI + '/' + URSI + '_' + testName + '-cov.fif', noiseCov)`

error:
`mne.write_cov(subjectDirectory + URSI + '/' + URSI + '_' + testName + '-cov.fif', noiseCov)`
`File "/usr/local/lib/python2.7/dist-packages/mne/cov.py", line 1179, in write_cov`
`  cov.save(fname)`
`File "/usr/local/lib/python2.7/dist-packages/mne/cov.py", line 152, in save`
`  raise inst`
`TypeError: list indices must be integers, not str`

Eric89GXL, 2017-01-31:
That's a weird one. Can you try writing the covariance to HDF5 with `mne.externals.h5io.write_hdf5('test.h5', noiseCov)` and uploading the result somewhere? If that works it's an easy way to debug and figure out what's wrong with the object.

Are you on latest `master` or 0.13?

WilliamCollishaw, 2017-01-31:
When trying to write hdf5 i got the following:

`Traceback (most recent call last):`
  `File "pythonScripts/pipeline.py", line 410, in <module>`
    `main()`
 `File "pythonScripts/pipeline.py", line 349, in main`
   `mne.externals.h5io.write_hdf5('test.h5', noiseCov)`
  `File "/usr/local/lib/python2.7/dist-packages/mne/externals/h5io/_h5io.py", line 106, in write_hdf5
    cleanup_data=cleanup_data, slash=slash, title=title)`
  `File "/usr/local/lib/python2.7/dist-packages/mne/externals/h5io/_h5io.py", line 137, in _triage_write
    where + '["%s"]' % key, cleanup_data=cleanup_data, slash=slash)`
  `File "/usr/local/lib/python2.7/dist-packages/mne/externals/h5io/_h5io.py", line 206, in _triage_write
    raise TypeError(err_str)`
`TypeError: unsupported type <class 'mne.cov._ShrunkCovariance'> (in <class 'mne.cov.Covariance'>["estimator"])`

I am using 0.14.git

Eric89GXL, 2017-01-31:
Okay let's try something else. I just pushed a commit that should give us a better traceback. Can you update to latest master, try running your code again, and paste the new, improved traceback?

WilliamCollishaw, 2017-01-31:
pretty sure this is exactly the same

`File "pipeline.py", line 413, in <module>`
   `main()`
  `File "pipeline.py", line 352, in main`
    `mne.externals.h5io.write_hdf5('test.h5', noiseCov)`
  `File "/na/homes/wcollishaw/pythonScripts/mne-python/mne/externals/h5io/_h5io.py", line 106, in write_hdf5`
    `cleanup_data=cleanup_data, slash=slash, title=title)`
  `File "/na/homes/wcollishaw/pythonScripts/mne-python/mne/externals/h5io/_h5io.py", line 137, in _triage_write`
    `where + '["%s"]' % key, cleanup_data=cleanup_data, slash=slash)`
  `File "/na/homes/wcollishaw/pythonScripts/mne-python/mne/externals/h5io/_h5io.py", line 206, in _triage_write`
    `raise TypeError(err_str)`
`TypeError: unsupported type <class 'mne.cov._ShrunkCovariance'> (in <class 'mne.cov.Covariance'>["estimator"])`

Eric89GXL, 2017-01-31:
Ahh sorry, I mean retry the `cov.save`

Eric89GXL, 2017-01-31:
... or `write_cov`

WilliamCollishaw, 2017-01-31:
`Traceback (most recent call last):`
  `File "pipeline.py", line 414, in <module>`
    `main()`
 `File "pipeline.py", line 353, in main`
    `mne.write_cov(subjectDirectory + URSI + '/' + URSI + '_' + testName + '-cov.fif', noiseCov)`
  `File "/na/homes/wcollishaw/pythonScripts/mne-python/mne/cov.py", line 1189, in write_cov`
    `cov.save(fname)`
 `File "/na/homes/wcollishaw/pythonScripts/mne-python/mne/cov.py", line 147, in save`
    `_write_cov(fid, self)`
  `File "/na/homes/wcollishaw/pythonScripts/mne-python/mne/cov.py", line 1805, in _write_cov`
    _write_proj(fid, cov['projs'])`
  `File "/na/homes/wcollishaw/pythonScripts/mne-python/mne/io/proj.py", line 405, in _write_proj`
    `write_int(fid, FIFF.FIFF_NCHAN, proj['data']['ncol'])`
`TypeError: list indices must be integers, not str`

Eric89GXL, 2017-01-31:
So there is something weird about your `cov['projs']`, which suggests there is something weird about `epochs.info['projs']`. I can get something similar with a nonsensical call:
```
>>> mne.write_proj('foo', 'bar')
<stdin>:1: RuntimeWarning: This filename (foo) does not conform to MNE naming conventions. All projection files should end with -proj.fif or -proj.fif.gz
This filename (foo) does not conform to MNE naming conventions. All projection files should end with -proj.fif or -proj.fif.gz
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "mne/proj.py", line 66, in write_proj
    io.proj._write_proj(fid, projs)
  File "mne/io/proj.py", line 405, in _write_proj
    write_int(fid, FIFF.FIFF_NCHAN, proj['data']['ncol'])
TypeError: string indices must be integers, not str
```
What is your `epochs.info['projs']`?

Do you get an error with `mne.write_proj('test-proj.fif', epochs.info['projs']`)? If so, how did you create your projectors? What raw data format are you using (FIF, CTF, etc.)?

Eric89GXL, 2017-01-31:
... oh I guess there might be something weird about `projList`. What is `projList`? Can you successfully do `mne.write_projs('test-proj.fif', projList)`?

WilliamCollishaw, 2017-01-31:
When trying to write projList i get the same error
projList is a list of projectors
`projList = []`
`projList.append(mne.read_proj(file))`

I tried to write_proj for a single projector and it worked.

in the documentation it states:
projs : list of Projection | None (default None)
List of projectors to use in covariance calculation, or None to indicate that the projectors from the epochs should be inherited. If None, then projectors from all epochs must match.

projs = list of projections.

Am I using it incorrectly?

Eric89GXL, 2017-01-31:
You want a list of projectors, so `projList.extend(mne.read_proj(...))` because `read_proj` returns a list itself. What you currently create is a list containing a list containing projectors.

Eric89GXL, 2017-01-31:
(we should probably add a check for this in the code to prevent this problem)

WilliamCollishaw, 2017-01-31:
So, I had a list of list of projectors, when i needed a list of projectors. I love data structures!

Using extend I was able to write_proj, will try recalculate the noise_cov and let you know my progress.

Thanks for all the prompt responses!

Eric89GXL, 2017-01-31:
No problem, fingers crossed it works this time...

WilliamCollishaw, 2017-01-31:
It worked! Thank you for the help!

Eric89GXL, 2017-01-31:
No problem. I'll re-title this to make it clear that we should raise an error if `projs_list` is not a list of projs

PR #3955 (2017-02-02): MRG: Check projectors
Closes #3945.

Checks that `projs` is actually valid when using `compute_covariance`. Would have saved @WilliamCollishaw some pain :)
closed 2017-02-02 by jaeilepp