mne-tools/mne-python #3951: Maxwell filter scaling.

jaeilepp, 2017-02-01:
When I use ``st_only=True`` the scaling on mag channels seems off. Script to reproduce.

```
import mne
import os.path as op
from mne.datasets import testing

data_path = testing.data_path(download=False)
sss_path = op.join(data_path, 'SSS')
fine_cal_fname = op.join(sss_path, 'sss_cal_3053.dat')
ctc_fname = op.join(sss_path, 'ct_sparse.fif')
pre = op.join(sss_path, 'test_move_anon_')
raw_fname = pre + 'raw.fif'
raw = mne.io.Raw(raw_fname, allow_maxshield=True)

raw_sss = mne.preprocessing.maxwell_filter(raw, calibration=fine_cal_fname,
                                           st_duration=2, st_only=True,
                                           cross_talk=ctc_fname)
raw_sss.plot()
```
I also see similar looking issues with reconstructed bad grad channels on [multi-subject, multi-modal human neuroimaging dataset](https://openfmri.org/dataset/ds000117/) subject 10, but I'm unable to reproduce it with the testing dataset.

Eric89GXL, 2017-02-01:
Hmm... I think that within any given window, the result actually looks reasonable. And tSSS is reducing power, as expected:
```
import mne
import os.path as op
from mne.datasets import testing
import numpy as np
from nose.tools import assert_true

data_path = testing.data_path(download=False)
sss_path = op.join(data_path, 'SSS')
fine_cal_fname = op.join(sss_path, 'sss_cal_3053.dat')
ctc_fname = op.join(sss_path, 'ct_sparse.fif')
pre = op.join(sss_path, 'test_move_anon_')
raw_fname = pre + 'raw.fif'
raw = mne.io.Raw(raw_fname, allow_maxshield='yes', preload=True)

raw_sss = mne.preprocessing.maxwell_filter(raw, calibration=fine_cal_fname,
                                           st_duration=2, st_only=True,
                                           cross_talk=ctc_fname)
raw.pick_types(meg='mag')
raw_sss.pick_types(meg='mag')
raw_sss.plot()
assert_true((np.linalg.norm(raw_sss[:][0], axis=-1) <
             np.linalg.norm(raw[:][0], axis=-1)).all())
```
But window-by-window the projection changes, so you get over multiple windows an up-down sort of behavior, which I assume is what you're pointing out? This might be fixed by doing overlap-add windowing, which is something I've discussed previously with Samu. I hope to get to it soon.

Eric89GXL, 2017-02-01:
Oh wait... I think there might also something odd going on due to the DC value in the signal. I'll keep digging.

Eric89GXL, 2017-02-01:
BTW anyone to lazy to run code, here's an example before/after tSSS, the 2-second-ness of the data processing is visible:
[image]
[image]

jaeilepp, 2017-02-02:
The bad channel reconstruction was fixed by setting ``origin`` and ``destination`` manually, but the mag channels are still noisy with ``st_only=True``. On multimodal dataset it's even more prominent.
[image]

Eric89GXL, 2017-02-02:
Can you paste the updated command lines so I can look at the same thing
locally?

jaeilepp, 2017-02-02:
That is from the multimodal dataset sub010. So just
```
raw = mne.preprocessing.maxwell_filter(raw, calibration=cal,
                                                   cross_talk=ctc,
                                                   st_duration=10.,
                                                   st_only=True,
                                                   origin=(0., 0., 0.04),
                                                   destination=destination)
```

Eric89GXL, 2017-02-02:
what is `destination` and which run is that?

Eric89GXL, 2017-02-02:
actually I think `destination` shouldn't matter for this, and I guess I can check any run and they'll probably display this behavior

Eric89GXL, 2017-02-02:
The original raw data with no projectors looks about the same. `st_only` removes projectors, so it will look different unless you disable proj.

I guess in `st_only` mode we don't need to remove projectors, since tSSS by itself doesn't/shouldn't change the spatial configuration of the data.

PR #3959 (2017-02-03): MRG+1: Fix projectors when st_only=True
Closes #3951.

No need to remove projectors when using `st_only=True`.

Ready for review/merge from my end.
closed 2017-02-03 by agramfort