mne-tools/mne-python #3953: Read in EGI GPS Digitization Coordinates

mattboggess, 2017-02-01:
To the best of my knowledge, MNE does not currently support reading in digitization files using EGI's GPS digitization process. An example file can be found [here](https://drive.google.com/file/d/0B-D6HZYNzLixTmowV19reWtTQWs/view?usp=sharing). I was hoping to integrate this functionality into MNE. I'd be willing to put in time and effort to do so, but was hoping for guidance on best approach for this integration. Thanks!

Eric89GXL, 2017-02-01:
For now you'd want to modify the `read_dig_montage` function / `DigMontage` class. Take a look at those and see if the necessary changes make sense

mattboggess, 2017-02-05:
I'm confused about the distinction between elp and hpi? Similarly, elp seems required to be set for both plotting the montage and transforming the coordinates though for the EGI montage only the fiducials and electrode positions are relevant. I'm not sure I understand what elp is supposed to contain?

Eric89GXL, 2017-02-06:
`hpi` are in MEG device space, `elp` are in digitizer space. So they're the same points in two different coordinate systems. They should only be necessary for determining an MEG<->head transformation

Eric89GXL, 2017-02-20:
I think this one was solved by PR

closed 2017-02-20 by Eric89GXL