mne-tools/mne-python #3956: ENH: gui coregistration

kingjr, 2017-02-01:
I don't seem to be able to modify the fiducials manually (i.e. by clicking on the surface) on `mne.gui.coregistration`. I think I used to be able to?..

Eric89GXL, 2017-02-01:
In case it's not possible yet (a la `mne_analyze), it looks like `mayavi` makes it fairly easy to get back to vertex numbers:

http://docs.enthought.com/mayavi/mayavi/auto/example_pick_on_surface.html

christianbrodbeck, 2017-02-01:
Even if you unlock them? Then it would be a new issue.

kingjr, 2017-02-01:
yes, even when I unlock them.. I tried with two different computers, one ubuntu one mac, both using CTF data

christianbrodbeck, 2017-02-01:
Are you talking about the MRI fiducials? Those work for me (macOS), also with mayavi 4.5.

Head Shape points could never be edited in the GUI.

christianbrodbeck, 2017-02-01:
(and they can be edited before specifying MEG/digitizer data, so that it is CTF data should not matter)

christianbrodbeck, 2017-02-01:
Also no error in the Terminal? If you can share the MRI subject I would see if I can modify them on my system.

kingjr, 2017-02-01:
> Are you talking about the MRI fiducials?

yes

I tried with freesurfer fsaverage too, same problem.

I encounter multiple errors, first pop up:

```
No high resolution head model was found for subject sub01, using standard head instead. In order to generate a high resolution head model, run:

    $ mne make_scalp_surfaces -s sub01
```

then another one:
```
No high resolution head model was found for subject sub01, using standard head instead. In order to generate a high resolution head model, run:

    $ mne make_scalp_surfaces -s sub01
```

and the following error in terminal:

```
Exception occurred in traits notification handler for object: <mayavi.tools.mlab_scene_model.MlabSceneModel object at 0x7f7f81466410>, trait: activated, old value: <undefined>, new value: True
Traceback (most recent call last):
  File "/home/jrking/anaconda/lib/python2.7/site-packages/traits/trait_notifiers.py", line 520, in _dispatch_change_event
    self.dispatch( handler, *args )
  File "/home/jrking/anaconda/lib/python2.7/site-packages/traits/trait_notifiers.py", line 483, in dispatch
    handler( *args )
  File "/home/jrking/git/mne-python/mne/gui/_coreg_gui.py", line 1335, in _init_plot
    self.model.hsp.sync_trait('points', p, mutual=False)
  File "/home/jrking/anaconda/lib/python2.7/site-packages/traits/has_traits.py", line 2893, in sync_trait
    setattr( object, alias, getattr( self, trait_name ) )
  File "/home/jrking/anaconda/lib/python2.7/site-packages/traits/trait_numeric.py", line 194, in validate
    self.error( object, name, value )
  File "/home/jrking/anaconda/lib/python2.7/site-packages/traits/trait_handlers.py", line 169, in error
    value )
TraitError: The 'points' trait of a PointObject instance must be an array of float64 values with shape ('*', 3), but a value of array([], dtype=float64) <type 'numpy.ndarray'> was specified.
```

christianbrodbeck, 2017-02-01:
To high res head warnings are expected because fsaverage does not have a high res head; you could suppress them with the `--low-res-head` option.

The second error looks like it's something about the data. Maybe you do have a corrupted fiducials file in your subject's directory? The GUI would try to load it automatically, and maybe that error then stops it from updating? If you have a fiducials file, could you move it out of the directory and try?

christianbrodbeck, 2017-02-01:
Although I just noticed it occurs on `self.model.hsp.sync_trait` – does the error also occur when you load no MEG data?

jaeilepp, 2017-02-02:
>  I tried with two different computers, one ubuntu one mac, both using CTF data

Could this be related to https://github.com/mne-tools/mne-python/pull/3922? Currently CTF conversion sets the fiducials to HPI points so the files probably don't have the fiducials in them. Somehow mne_analyze allows the coregistration without the fiducials.

christianbrodbeck, 2017-02-02:
Yes I seem to remember there was something about a corrupted fiducials file a while ago that's why I suggested moving it away (if there is one); if that's it we should modify the GUI so that it catches the error.

avkasdan, 2017-02-07:
(This is JR on another computer):

We encounter the error on multiple CTF files. When launching mne.gui.coregistration without the raw instance fname specified we get : 

```Error reading fiducials from ATE26_TvsOpred_20150512_03.ds: [Errno 21] Is a directory: u'/Users/poeppellab/Documents/ATE26_TvsOpred_20150512_03.ds' (See terminal for more information```

In the terminal we get

```
2017-02-07 11:50:00.897 python[62689:3000545] modalSession has been exited prematurely - check for a reentrant call to endModalSession:
setting roll
2017-02-07 11:50:15.120 python[62689:3000545] Unable to simultaneously satisfy constraints:
(
    <NSAutoresizingMaskLayoutConstraint:0x12d179230 h=--& v=--& V:[FI_TBrowserBackgroundView:0x12e47c050(0)]>,
    <NSAutoresizingMaskLayoutConstraint:0x12d1be4c0 h=-&- v=-&- V:|-(0)-[FI_TListScrollView:0x12ac79fc0]   (Names: '|':FI_TBrowserBackgroundView:0x12e47c050 )>,
    <NSAutoresizingMaskLayoutConstraint:0x12d1be470 h=-&- v=-&- V:[FI_TListScrollView:0x12ac79fc0]-(0)-|   (Names: '|':FI_TBrowserBackgroundView:0x12e47c050 )>,
    <NSAutoresizingMaskLayoutConstraint:0x12b9af800 h=-&- v=-&- V:[NSClipView:0x1298d3210]-(15)-|   (Names: '|':FI_TListScrollView:0x12ac79fc0 )>,
    <NSAutoresizingMaskLayoutConstraint:0x12b9af7b0 h=-&- v=-&- V:|-(0)-[NSClipView:0x1298d3210]   (Names: '|':FI_TListScrollView:0x12ac79fc0 )>
)

Will attempt to recover by breaking constraint 
<NSAutoresizingMaskLayoutConstraint:0x12b9af800 h=-&- v=-&- V:[NSClipView:0x1298d3210]-(15)-|   (Names: '|':FI_TListScrollView:0x12ac79fc0 )>

Set the NSUserDefault NSConstraintBasedLayoutVisualizeMutuallyExclusiveConstraints to YES to have -[NSWindow visualizeConstraints:] automatically called when this happens.  And/or, break on objc_exception_throw to catch this in the debugger.
```

However the fiducials are now interactive.

christianbrodbeck, 2017-02-07:
Theres something wrong. Fiducials filenames should never be anything with an extension other than `*.fif`. Can you check the output of `mne.coreg._find_fiducials_files(subject, subjects_dir)`?

Eric89GXL, 2017-02-16:
> Fiducials filenames should never be anything with an extension other than *.fif

Hmm... we might want to change this. We could relax the requirement to be anything that can be read with a `mne.io.read_raw_*` function. Some readers, like CTF, should populate `info['dig']` properly.

But in the meantime, @kingjr you can probably work around this by doing something like:
```
raw = mne.io.read_raw_ctf(...)
mne.io.write_dig('whatever-dig.fif', raw.info['dig'])
```
And then pass that FIF filename to the coreg GUI. Can you try it?

Eric89GXL, 2017-02-28:
The clicking issue has been fixed. I've added the non-FIF idea to #3934

closed 2017-02-28 by Eric89GXL