mne-tools/mne-python #3957: reading eeglab files with channels missing position

mmagnuski, 2017-02-01:
I have many eeglab files that contain at least one channel that has undefined position (for example biosemi EXG channels) - the way `mne.io.eeglab._get_info` works currently is to ignore all channel positions if at least one channel has no specified position. 
This behavior originates [here](https://github.com/mne-tools/mne-python/blob/master/mne/io/eeglab/eeglab.py#L81) - `np.unique` of x, y and z position is checked and if the size of the unique is == 1 the channel positions are ignored. In the case of no position for a given channel - `_to_loc` which is used to get channel positions returns 0 for x, y and z - in which case unique is of size 1 and all channel positions are ignored.
I think it would be better if `_to_loc` returned `np.nan` when given value was not present and channels with at least one `np.nan` in position were omitted when building Montage. I can make a PR if you agree.

jasmainak, 2017-02-01:
I don't know what is the consequence of having `NaN`s in the `info`. Perhaps @agramfort or @Eric89GXL can have better opinion here

mmagnuski, 2017-02-01:
nans would be omitted and not included in montage - so no nans in info. In the current setup you can get 0 if there is no position info or if the position is 0.

mmagnuski, 2017-02-01:
I have changed that part of code so it may be easier to just show it instead of explaining :) :#3958

prefrontalvortex, 2017-02-02:
Is this the error you are seeing? 

```
/lib/python3.5/site-packages/mne/io/eeglab/eeglab.py in _get_info(eeg, montage, eog)
     75         for chanloc in eeg.chanlocs:
     76             ch_names.append(chanloc.labels)
---> 77             loc_x = _to_loc(chanloc.X)
     78             loc_y = _to_loc(chanloc.Y)
     79             loc_z = _to_loc(chanloc.Z)

AttributeError: 'mat_struct' object has no attribute 'X'
```

mmagnuski, 2017-02-02:
@prefrontalvortex no, that seems to be another issue - how does your EEG.chanlocs look (in matlab for example)?

prefrontalvortex, 2017-02-02:
Empty. 

`[]`

Looks like it's never dealt with a case where the chanlocs struct is empty.

prefrontalvortex, 2017-02-02:
I'm a "beg forgiveness > ask permission" kind of guy, so I did this and was able to get rolling with the data:
```
    try:
        if len(eeg.chanlocs) > 0:
            ch_names, pos = list(), list()
            kind = 'user_defined'
            selection = np.arange(len(eeg.chanlocs))
            locs_available = True
            for chanloc in eeg.chanlocs:
                ch_names.append(chanloc.labels)
                loc_x = _to_loc(chanloc.X)
                loc_y = _to_loc(chanloc.Y)
                loc_z = _to_loc(chanloc.Z)
                locs = np.r_[-loc_y, loc_x, loc_z]
                if np.unique(locs).size == 1:
                    locs_available = False
                pos.append(locs)
            if locs_available:
                montage = Montage(np.array(pos), ch_names, kind, selection)
        elif isinstance(montage, string_types):
            path = op.dirname(montage)
        else:  # if eeg.chanlocs is empty, we still need default chan names
            ch_names = ["EEG %03d" % ii for ii in range(eeg.nbchan)]
    except AttributeError as err:
        print('Warning, failed to parse channel location or something: {}'.format(err))

```

If all this function is doing is parsing meta-data, it might be better just to try/except/warn instead of either `if len(eeg.chanlocs) > 0:` or ` for chanloc in eeg.chanlocs:`

jasmainak, 2017-02-03:
The problem with a try-catch construct is that many other errors will pass silently. You don't want that ...

mmagnuski, 2017-02-03:
Ok, I'll fix that in the open PR - that requires only a small change.

mmagnuski, 2017-02-03:
That should work now - I've added checks for X, Y and Z fields and a relevant test that passes.

PR #3958 (2017-02-07): [MRG +1]: construct Montage out of channels with position when reading eeglab files
fixes #3957
closed 2017-02-07 by jasmainak