mne-tools/mne-python #3960: Ground truth timeseries from simulation?

jhouck, 2017-02-02:
If I've created a source estimate with simulated data (as in the plot_simulate_raw_data example), what would be the most straightforward way to extract the full timeseries before it's interpolated in simulate_raw?  It looks like it would be a little more involved than concatenating multiple copies of stc.data from the simulated source estimate to create a couple of minutes of data.  Or is it?  The eventual goal is to apply the same operations to the 'ground truth' timeseries(es) and to the reconstructed simulated data.

Eric89GXL, 2017-02-02:
It should just be the concatenation. You can verify it in many ways, e.g.,
by checking that the event times correspond to t=0 in your repeating STC,
or that (in the noise free case) any given repeat is a linear combination
of your N original STC time courses. Sinusoids or temporal delta functions
are good for the latter

jhouck, 2017-02-03:
Excellent - thanks!

closed 2017-02-03 by Eric89GXL