mne-tools/mne-python #3963: How to define channel types on the fly

cmmoenne, 2017-02-03:
In my lab we use multiple synced signals (EEG, eye-tracking, ...), and one of them is information of spatial markers (ndigital optotrak). I create RawArrays from the spatial data with 'misc' channel types. The problem is that every time I want to plot the evoked data a __ValueError: Non-data channel picked.__ is raised. My temporary (lazy) solution is to change the channels types to a valid one (as those defined in _DATA_CH_TYPES_SPLIT), but it is far from a good solution. So it would be nice if I could define a channel type (with its units to plot, etc) when using other ___non-standard___ data types (e.g. eye-tracking/mouse-tracking channels in pixels or mm, spatial markers in mm, among others) while profiting of this great software.

agramfort, 2017-02-04:
you mean evoked.plot crashes when you try to plot a misc channel?

cmmoenne, 2017-02-04:
I don't know if it is a crash or expected behavior. For example, in [plot_compare_evokeds line 1475](https://github.com/mne-tools/mne-python/blob/master/mne/viz/evoked.py#L1475) If any channel doesn't belong to _DATA_CH_TYPES_SPLIT (defined in io.pick [here](https://github.com/mne-tools/mne-python/blob/master/mne/io/pick.py#L715)) the value error is raised.

That is not the case with plot_evoked as misc channels are defined as valid [here](https://github.com/mne-tools/mne-python/blob/master/mne/viz/evoked.py#L202). So my apologies for not specifying the function with the problem.

agramfort, 2017-02-05:
if evoked.plot supports misc then it should also be the case for the
plot_compare_evokeds function

can you open a PR to fix it yourself?

cmmoenne, 2017-02-05:
I open PR #3965 with fixes for plot_compare_evoked. So I close this issue now.

PR #3965 (2017-03-13): MRG: Fix plot_compare_evokeds to support 'misc' channel type
Fixes #3963 

to allow use 'misc' channels in plot_compare_evokeds. I replace _DATA_CH_TYPES_SPLIT (defined as ['mag', 'grad', 'eeg', 'seeg', 'ecog', 'hbo', 'hbr'] [see here](https://github.com/mne-tools/mne-python/blob/master/mne/io/pick.py#L715)) with the same valid types defined in [_plot_evoked](https://github.com/mne-tools/mne-python/blob/master/mne/viz/evoked.py#L200).
closed 2017-02-05 by cmmoenne