mne-tools/mne-python #3969: accessing fieldtrip buffer (MATLAB) in MNE-python ?

nbara, 2017-02-06:
Hi all,

Apologies if this is an obvious question but I'm having trouble finding the answer.

Let's assume I start a fieldtrip buffer from Matlab:

```
cfg                = [];
% cfg.channel    = 1:4;                      % list with channel "names"
cfg.blocksize    = 0.5;                       % seconds
cfg.fsample      = 64;                        % sampling frequency, Hz
cfg.target.dataset = 'buffer://localhost:1972';   % where to write the data
ft_realtime_signalproxy(cfg)
```

Should it be possible to access it in python (code below from the  MNE's [examples](https://github.com/mne-tools/mne-python/blob/master/examples/realtime/ftclient_rt_average.py)?)

```
with FieldTripClient(host='localhost', port=1972,tmax=150, wait_max=10) as rt_client:

    # get measurement info guessed by MNE-Python
    raw_info = rt_client.get_measurement_info()

```

I get the following error:

```
  File "/Users/nbara/Toolboxes/mne-python/mne/realtime/fieldtrip_client.py", line 158, in _guess_measurement_info
    this_info['logno'] = int(re.findall('[^\W\d_]+|\d+', ch)[-1])

ValueError: invalid literal for int() with base 10: 'Fpz'
```

 `raw_info` contains 0 channels

Is this a bug or am I missing something obvious?

Eric89GXL, 2017-02-06:
That's a bug. It's trying to guess the measurement info from the channel name and failing. I'm curious, what  system are you streaming data from? It's possible people have only done Neuromag systems so far, in which case we wouldn't have hit this.

@jasmainak do you have time to look?

nbara, 2017-02-07:
Hi Eric,

At the moment I was just trying to generate a dummy EEG signal with the above matlab code, but the next step is to use the FT buffer with a Biosemi device.

From what I can gather, the code in `fieldtrip_client.py` seems to be tailored for Neuromag files indeed, hence the error.

nbara, 2017-02-07:
If it helps, I've managed to circumvent all the data checks (which are all Neuromag-centered) with some quick and ugly code:

In MATLAB:
```
% generate dummy data
cfg            = [];
cfg.channel    = 1:2;   % create 2 channels// default 10-20 names will fail in MNE
cfg.blocksize  = 0.5;   % seconds
cfg.fsample    = 64;    % sampling frequency, Hz
cfg.target.dataset = 'buffer://localhost:1972';   % where to write the data
ft_realtime_signalproxy(cfg); 
```

In Python, I'm basically overriding `FieldTripClient. _guess_measurement_info(self)`

```
with FieldTripClient(host='localhost', port=1972,tmax=150, wait_max=10) as rt_client:

    # get measurement info guessed by MNE-Python
    raw_info = rt_client.get_measurement_info()
    
    # Manually replace
    raw_info['ch_names'] = (['Oz', 'Pz'])  # can be any name
    raw_info['nchan']    = 2;  # FIFFV_MNE_EEG
    raw_info['projs']    = list() 
    raw_info['chs'].append({'ch_name':'Oz','range':1.0,'cal':1.0,'unit_mul':0,'kind':2})
    raw_info['chs'].append({'ch_name':'Pz','range':1.0,'cal':1.0,'unit_mul':0,'kind':2})
```    

It's quite probable that the above code will break with real data though.

Eric89GXL, 2017-02-07:
Yeah that looks like the right general direction for the fix, but we'll have to make something more general. Do you have time to try a complete fix and open a PR?

nbara, 2017-02-07:
I'd be happy to try, but I would need some guidance (I'm still a python noob!)

jasmainak, 2017-02-07:
@nbara sorry for the late response.

Did you try what I suggest [here](https://github.com/mne-tools/mne-python/pull/3620#discussion_r80743202). This is really the only foolproof way to make sure things work. Let me look at your solution ...

jasmainak, 2017-02-07:
@Eric89GXL what is `info['logno']` [supposed to store](https://github.com/mne-tools/mne-python/blob/master/mne/realtime/fieldtrip_client.py#L158)? I suppose we could do away with the regexp and simply replace it with the iteration number?

nbara, 2017-02-08:
> Did you try what I suggest here. This is really the only foolproof way to make sure things work.

Is there an easier way to populate/initialise the `info` structure than copying it from a dummy file on disk?

jasmainak, 2017-02-08:
> Is there an easier way to populate/initialise the info structure than copying it from a dummy file on disk?

Maybe you can use [`create_info`](http://martinos.org/mne/stable/generated/mne.create_info.html) ?

> Besides, shouldn't we assume that FieldtripClient should 'just work' with a fieldtrip buffer?

The problem is that critical information such as channel type are not available in the header that fieldtrip buffer transmits ...

nbara, 2017-02-08:
> Maybe you can use create_info ?

Yes, that does it nicely. For instance with 4 channels 

```
montage = mne.channels.read_montage('biosemi64') # or whatever montage you wish
info    = mne.create_info(montage.ch_names[:4],256,ch_types='eeg',montage=montage)

# Open FT buffer
with FieldTripClient(info=info, host='localhost', port=1972,
                     tmax=150, wait_max=10, verbose=0) as rt_client:

    # SOME CODE
```

Thanks @jasmainak , closing now :)

closed 2017-02-08 by nbara