mne-tools/mne-python #3970: Unit of measure for raw data in mne.io.RawArray

kaczmarj, 2017-02-06:
I am trying to load EEG data from a `numpy` array into an `mne.io.RawArray` object. What is the correct unit of measure for the data in the array? Volts? I looked through the source for `mne.create_info`, and it looks like the unit should be volts. Can anyone confirm?

Thank you.

Eric89GXL, 2017-02-06:
Yes volts. Do you have time to open a PR to clarify the documentation for
the next person?

kaczmarj, 2017-02-07:
I could do that. Because a class like `RawArray` can accept more than just EEG data, how would you suggest I add the proper unit of measure to the documentation?

Eric89GXL, 2017-02-07:
Probably a bulleted list. To keep it slightly shorter, you can organize it by type:
```
* V: eeg, eog, seeg, emg, ecg, bio, ecog
* T: mag
* T/m: grad
* M: hbo, hbr
* Am: dipole
* AU: misc
```
And I guess it should probably go in `create_info`, too / instead of `RawArray`?

kaczmarj, 2017-02-07:
Thanks. I think it would make sense to include that list in the docstring of each `*Array` class, and in the notes of `create_info`.

Eric89GXL, 2017-02-07:
Sounds good to me

agramfort, 2017-02-07:
this tutorial should be edited too:

http://martinos.org/mne/stable/auto_tutorials/plot_creating_data_structures.html

kaczmarj, 2017-02-07:
To edit that tutorial, would I edit https://github.com/mne-tools/mne-python/blob/master/tutorials/plot_creating_data_structures.py ?

agramfort, 2017-02-07:
yes exactly

closed 2017-02-07 by kaczmarj