mne-tools/mne-python #3971: problems with suggested info structure in ECOG example

MartinPerez, 2017-02-07:
The example for an Ecog data suggest using the label 'ecog' when creating the info structure, but later when trying to use the plotting functions this leads to very annoying errors that cant be corrected with any possible parameters. For example plot_psd of the raw object do not accept a custom layout but will eventually call find_layout, that returns None, inside _plot_lines. Then it tries to access the channel names from the layout instead of the info structure returning an Exception.

/volatile/martin_local/OherEnvs/blackboard/local/lib/python2.7/site-packages/mne-0.14.dev0-py2.7.egg/mne/viz/evoked.pyc in _plot_lines(data, info, picks, fig, axes, spatial_colors, unit, units, scalings, hline, gfp, types, zorder, xlim, ylim, times, bad_ch_idx, titles, ch_types_used, selectable, psd, line_alpha)
    340                     used_nm = np.array(_clean_names(info['ch_names']))[idx]
    341                     names = np.asarray([name for name in used_nm
--> 342                                         if name in layout.names])
    343                     name_idx = [layout.names.index(name) for name in names]
    344                     if len(name_idx) < len(chs):

AttributeError: 'NoneType' object has no attribute 'names'

For the moment I am bypassing these problems by simply considering the electrodes as 'mag' instead. I am new to MNE so I am not sure I can bring forward a good proposal to fix this but I am happy to help if I can. What do you recommend me to do at the moment?

MartinPerez, 2017-02-07:
Naively I would ask why a custom Layout would not form part directly of the Info object, since a Montage is given to it anyway with all the type of information I would use for a custom layout.

Eric89GXL, 2017-02-07:
> The example for an Ecog data suggest using the label 'ecog' when creating the info structure

Yes this is the right thing to do...

> but later when trying to use the plotting functions this leads to very annoying errors that cant be corrected with any possible parameters

...but there might be issues. Let's fix them.

> What do you recommend me to do at the moment?

First step is to make a minimal example e.g. using `RawArray` that shows the problem. Can you take care of that and post the code here?

The next step is to turn that example into a failing test (test-driven development!) in some suitable place like `mne/io/array/tests/test_array.py`, and open a PR to add the test. The final step would be to fix the error. Knowing the layout code, I think this one might be tough so maybe @jaeilepp or someone else could have a look?

Eric89GXL, 2017-02-07:
> Naively I would ask why a custom Layout would not form part of the Info object, since a Montage is given to it anyway with all the type of information I would use for a custom layout.

Right -- the montage should set the channel positions, which should then be used to set the layout. So I think we have two bugs actually: 1) plotting should not fail if there is no layout available, and 2) the layout should be able to be created on the fly from `info` but for some reason it isn't.

MartinPerez, 2017-02-07:
@Eric89GXL thanks for your answers. I will be happy to assist with the minimal example/test. I am not familiar with any of the objects, so I am not sure I will bring good insights to fix the bugs. Its actually my first time using MNE. Nonetheless I can give a hand to anyone that points out the precise changes to attempt.

One last question is that for my electrodes I have MNI coordinate positions. Which I provided to the DigitalMontage, but I found no way to clarify the coordinate system of my positions. I am assuming in a Layout it would be given by playing with the box and providing an appropriate background? plot_psd_topo of course breaks with the type of coordinates im providing. What do you suggest in that case?

Eric89GXL, 2017-02-07:
> One last question is that for my electrodes I have MNI coordinate positions.

Is it possible for you also to get the fiducials (LPA, RPA, Nasion) from your MRI? If so, then passing those to the `read_dig_montage` you can convert them all to the head coordinate system, and also have access to the head<->MRI transform, which is useful.

Your use case is actually something that we're planning to fix/clarify in the coming couple of months (#3699, #3893) because it should be easier.

MartinPerez, 2017-02-07:
>Is it possible for you also to get the fiducials (LPA, RPA, Nasion) from your MRI?

The dataset was given to me with that information only from a mix of places and times. Likely I wont be able to access further details. All the coordinates are in MNI space, since they are normalized already I hope there is workaround simply by playing with the Montage and/or Layout objects. I could PR an example at the end from this experience.

MartinPerez, 2017-02-08:
Here I post a simple example to arrive to the problem. Specifically that raw.plot_psd(average=False) tries to call a layout that can not be overridden as an argument of the function.

```
import mne
import numpy as np

n_elec = 10
ts_size = 10000
Fs = 512.
elec_labels = [str(i) for i in range(n_elec)]
elec_coords = np.random.randint(60, size=(n_elec, 3)).tolist()

electrode = np.random.rand(n_elec, ts_size)
dig_ch_pos = dict(zip(elec_labels, elec_coords))
mon = mne.channels.DigMontage(dig_ch_pos=dig_ch_pos)
info = mne.create_info(elec_labels, Fs, 'ecog', montage=mon)

raw = mne.io.RawArray(electrode, info)

raw.plot_psd(average=False)
```

MartinPerez, 2017-02-08:
Would you like me to PR it as a smoke test in mne/io/array/tests/test_array.py?

agramfort, 2017-02-08:
yes please do

thx

MartinPerez, 2017-02-08:
PR up

closed 2017-03-17 by agramfort