mne-tools/mne-python #3974: BUG: problem with mne.bem.make_watershed_bem

romquentin, 2017-02-07:
Hi all, 
I am trying to run `make_watershed_bem` and I got the following error.

```
Running mri_watershed for BEM segmentation with the following parameters:

SUBJECTS_DIR = /Users/quentinra/Desktop/MEGdata/WM1/subjects
SUBJECT = sub03_XATEOUZW_WM
Results dir = /Users/quentinra/Desktop/MEGdata/WM1/subjects/sub03_XATEOUZW_WM/bem/watershed

Running subprocess: mri_watershed -useSRAS -surf /Users/quentinra/Desktop/MEGdata/WM1/subjects/sub03_XATEOUZW_WM/bem/watershed/sub03_XATEOUZW_WM /Users/quentinra/Desktop/MEGdata/WM1/subjects/sub03_XATEOUZW_WM/mri/T1.mgz /Users/quentinra/Desktop/MEGdata/WM1/subjects/sub03_XATEOUZW_WM/bem/watershed/ws

Mode:          use surfaceRAS to save surface vertex positions
Mode:          Saving out BEM surfaces

*********************************************************
The input file is /Users/quentinra/Desktop/MEGdata/WM1/subjects/sub03_XATEOUZW_WM/mri/T1.mgz
The output file is /Users/quentinra/Desktop/MEGdata/WM1/subjects/sub03_XATEOUZW_WM/bem/watershed/ws
dyld: lazy symbol binding failed: Symbol not found: ___emutls_get_address
  Referenced from: /Applications/freesurfer/bin/../lib/gcc/lib/libgomp.1.dylib
  Expected in: /usr/lib/libSystem.B.dylib

dyld: Symbol not found: ___emutls_get_address
  Referenced from: /Applications/freesurfer/bin/../lib/gcc/lib/libgomp.1.dylib
  Expected in: /usr/lib/libSystem.B.dylib

('', '')
---------------------------------------------------------------------------
CalledProcessError                        Traceback (most recent call last)
<ipython-input-6-6453950d4fbb> in <module>()
      3                    overwrite=True, volume='T1', atlas=False,
      4 
----> 5                    gcaatlas=False, preflood=None, show=True)

/Applications/anaconda/lib/python2.7/site-packages/mne-python/mne/bem.pyc in make_watershed_bem(subject, subjects_dir, overwrite, volume, atlas, gcaatlas, preflood, show, verbose)

/Applications/anaconda/lib/python2.7/site-packages/mne-python/mne/utils.pyc in verbose(function, *args, **kwargs)
    707         with use_log_level(verbose_level):
    708             return function(*args, **kwargs)
--> 709     return function(*args, **kwargs)
    710 
    711 

/Applications/anaconda/lib/python2.7/site-packages/mne-python/mne/bem.pyc in make_watershed_bem(subject, subjects_dir, overwrite, volume, atlas, gcaatlas, preflood, show, verbose)
   1103                 'Results dir = %s\n' % (subjects_dir, subject, ws_dir))
   1104     os.makedirs(op.join(ws_dir, 'ws'))
-> 1105     run_subprocess(cmd, env=env, stdout=sys.stdout, stderr=sys.stderr)
   1106 
   1107     if op.isfile(T1_mgz):

/Applications/anaconda/lib/python2.7/site-packages/mne-python/mne/utils.pyc in run_subprocess(command, verbose, *args, **kwargs)

/Applications/anaconda/lib/python2.7/site-packages/mne-python/mne/utils.pyc in verbose(function, *args, **kwargs)
    707         with use_log_level(verbose_level):
    708             return function(*args, **kwargs)
--> 709     return function(*args, **kwargs)
    710 
    711 

/Applications/anaconda/lib/python2.7/site-packages/mne-python/mne/utils.pyc in run_subprocess(command, verbose, *args, **kwargs)
   1217         err_fun = subprocess.CalledProcessError.__init__
   1218         if 'output' in _get_args(err_fun):
-> 1219             raise subprocess.CalledProcessError(p.returncode, command, output)
   1220         else:
   1221             raise subprocess.CalledProcessError(p.returncode, command)

CalledProcessError: Command '['mri_watershed', '-useSRAS', '-surf', '/Users/quentinra/Desktop/MEGdata/WM1/subjects/sub03_XATEOUZW_WM/bem/watershed/sub03_XATEOUZW_WM', '/Users/quentinra/Desktop/MEGdata/WM1/subjects/sub03_XATEOUZW_WM/mri/T1.mgz', '/Users/quentinra/Desktop/MEGdata/WM1/subjects/sub03_XATEOUZW_WM/bem/watershed/ws']' returned non-zero exit status -5
```

Surprisingly, running directly in the shell 
`mri_watershed -useSRAS -surf /Users/quentinra/Desktop/MEGdata/WM1/subjects/sub03_XATEOUZW_WM/bem/watershed/sub03_XATEOUZW_WM /Users/quentinra/Desktop/MEGdata/WM1/subjects/sub03_XATEOUZW_WM/mri/T1.mgz /Users/quentinra/Desktop/MEGdata/WM1/subjects/sub03_XATEOUZW_WM/bem/watershed/ws` works without error.

I am running mne.014dev and freesurfer v6.

Any helps?

Thanks a lot

Romain

agramfort, 2017-02-09:
it's likely to be a conflict of dynamic libraries between python and
freesurfer

I doubt there is an easy fix for this.

did you manually edit the LD_LIBRARY_PATH or DYLD_LIBRARY_PATH on mac?

Eric89GXL, 2017-02-09:
We could implement the same fix that PySurfer uses, namely:

1. Require users to set `FREESURFER_HOME`
2. Don't require them to source the FreeSurfer setup script (thus avoiding potentially nasty lib conflicts)
3. When we go to make a FS call, we source the script in the `subprocess` script before making the call

romquentin, 2017-02-09:
Ok. I did not edit LD_LIBRARY_PATH or DYLD_LIBRARY_PATH. 
Not sure to follow Eric suggestions... In the meantime, I can run this command on bash. 
Thanks!!

Eric89GXL, 2017-02-16:
> Ok. I did not edit LD_LIBRARY_PATH or DYLD_LIBRARY_PATH.

Well assuming that you did run the FreeSurfer setup script in the terminal, it modifies these. So it could break something at the Python end. Thus why we should remove the requirement that people have done this prior to running `python`.

larsoner, 2019-08-20:
I think we've fixed this stuff recently with proper error messages, feel free to reopen if it's still happening

closed 2019-08-20 by larsoner