mne-tools/mne-python #3979: Question: apply gyroscope data to detect / remove artifacts?

ppasler, 2017-02-09:
I have gyroscope channels recorded with my eeg data (the device is able to record both).

Is there any built-in way to automatically detect (and remove) artifacts from head movement, based on the gyroscope data? Something similar to this example for eog / ecg: http://martinos.org/mne/dev/auto_tutorials/plot_artifacts_correction_ica.html

Thanks for any hints.

agramfort, 2017-02-10:
you're the first to ask so there is none :)

ppasler, 2017-02-10:
Too bad :)

I can see strong correlations between the amplitudes in the gyrochannels (head movement) and artifacts in the EEG signal.

I wonder if there is some way, to compare the ICA source channels with the gyro channel and find similarities. So if an ICA source channels fits the gyro channel, it may be removed?

agramfort, 2017-02-10:
sounds like a reasonable approach

dgwakeman, 2017-02-10:
it may also be easy to just try setting a hard reject limit on the gyroscope channels. That would seem to be the simpler solution. It also might make more sense, since the gyroscope channels are essentially a measurement of noise (there is of course an issue of how noisy those channels are).

ppasler, 2017-02-10:
@agramfort I also thought this could work :) But @dgwakeman is right. 

As I analyzed the signals, the head movement artifacts were visible in many ICA channels without a chance to zero all of them. I faced this problem before when trying to remove EOG artifacts with the corrmap algorithm, local maxima from a blink showed up in several ica source channels. I did not find a way (I don't think there is one), to adjust the ICA in a way that the blinks / head movements only show up in 1 ICA source channel.

I am now using the gyro information to reject epochs (as suggested). 
I calculate the signal energy and if it's above a threshold, I reject the epoch - this worked well and improved my results!

Eric89GXL, 2017-02-16:
Okay closing this for now, since it seems like there is a suitable way to do it already

closed 2017-02-16 by Eric89GXL

ppasler, 2017-02-17:
@Eric89GXL alright. Still I think it would be considerable to add a build-in solution for this to mne. Thanks anyway :)