mne-tools/mne-python #3981: API/BUG: empty list in mne.set_eeg_reference

dengemann, 2017-02-10:
`mne.set_eeg_reference` applies either the average reference or reference based on specified channels. But why the heck do we support passing an empty list, which does essentially nothing but setting `.info['custom_ref_applied']` to True. To me this is a confusing API, we should either set a reference or not. Calling `mne.set_eeg_reference([])` should not be allowed. Am I overlooking something?

Eric89GXL, 2017-02-13:
Agreed empty list should be an error, PR welcome

wmvanvliet, 2017-02-20:
The empty list use case was added as a way to specify that the data already has the proper reference and to leave it alone. Without it, MNE will automatically add an average reference at every turn.

wmvanvliet, 2017-02-20:
Any PR that removes this behaviour should **subsequently go through all the instances where an average reference is automatically added and change this too.**

Eric89GXL, 2017-02-21:
@dengemann are you convinced? Okay to close?

dengemann, 2017-02-21:
I am not yet fully convinced :) This is highly confusing to newcomers as I
have recently learned while looking at failing data analyses. We should
better document it at least and add some better verbosity. Or it should not
be an empty list but some string like "added" or "referenced", etc. I'm on
the other hand against thoughtless deprecation. Marijn what would you think
of printing an info message that epxplains what's happening?

wmvanvliet, 2017-02-21:
I'm curious: what are the users trying to accomplish with passing an empty list to the function, if they do not know that it disables any further automatic referencing?

dengemann, 2017-02-21:
They think [] or None are False and want to achieve average referencing.

agramfort, 2017-02-21:
@dengemann what you have in mind for a lest confusing API?

dengemann, 2017-02-21:
As I said, I have become very conservative on deprecations. Printing the
unlikely, the unexpected, would be one choice. But if I'd rewewrite it
again i'd use explicit string codes, 'average_ref', 'makred_as_ref' or so.

Eric89GXL, 2017-02-21:
So your vote is just to make documentation changes, then? If so please close

dengemann, 2017-02-21:
No, more then documentation. Strong verbosity is needed here if we want to
keep the API.

Eric89GXL, 2017-02-21:
Can you be more explicit? What is needed on top of / instead of #4004?

dengemann, 2017-02-21:
Sure, let me comment and review there and close here once I'm happy.

Eric89GXL, 2017-02-21:
Sounds good. If the changes seem outside the scope of #4004, we can also just add them as a to-do list to the top of this issue

dengemann, 2017-02-21:
Roger.

dengemann, 2017-02-22:
We can close this with #4004.

closed 2017-02-22 by Eric89GXL