mne-tools/mne-python #3985: Documentation improvements for the docathon?

choldgraf, 2017-02-13:
Hey all - as some of you guys may know, at BIDS Nelle V. and I are organizing a Docathon in early March (bids.berkeley.edu/docathon). It'll be a week where we try to get people to focus on improving / building / writing tools for their documentation. There will be a few remote docathons as well, and anyway we are encouraging anybody to participate via github pushes etc (we'll also have a slack channel), so you don't have to physically be in California to help out.

I think it'd be great to use this as an opportunity to improve some of the MNE documentation, so I wonder if anybody would be interested in spending some time on this during the week. If so, I can submit MNE as a project.

Can we come up with some "doable in a week" improvements to work on? Some possible ideas:

* Improve tutorials that cover the basics, if we're missing a topic somewhere
* Make sure we have examples / explanations of the more complicated algorithms in the package (e.g. I think there's an issue about explaining the BEM algos better in #3082)
* Add more "comprehensive" documentation about particular topics (e.g. right now we have only 2 tutorials in the multivariate statistics section including the one in the encoding model PR, perhaps we could add explanations about how MNE could be used with something like scikit-learn).

Let me know your thoughts! I'll probably be spending some of the week organizing things more generally, but can hopefully spend some time on MNE as well (most of the days we'll just spend working in general).

And if you're interested in the docathon more generally, I can chat more about that if anyone wants :)

kingjr, 2017-02-13:
I think we should merge the 'manual' and the 'tutorials'.

Eric89GXL, 2017-02-14:
Good targets would be:

* @choldgraf and @kingjr ideas for [updating the workflow](https://github.com/mne-tools/mne-python/issues/3669), especially if someone wants to make a client-side rendered workflow image :)
* @teonbrooks [small issues](#3082) you hit
* @dengemann [SNR doc](#1857)
* @dengemann [connectivity doc](#2847)
* and there is of course the [doc mega-issue](#1495)

Eric89GXL, 2017-02-14:
And:

* @christianbrodbeck [tutorial on coreg](#3358)

choldgraf, 2017-02-14:
@Eric89GXL it seems like most of those links just redirect back to this issue?

Those are great suggestions though. If we can get an idea for prioritizing these in some way (e.g. I think a tutorial on coregistration would be useful, especially if there could eventually be support for doing ecog coregistration using a CT scan or something)

claire-braboszcz, 2017-03-02:
I've seen on [Project](https://github.com/mne-tools/mne-python/projects/1) that @kingjr was thinking of writing a tutorial on time-frequency decoding. I'd be very interested in such a tutorial that would complement nicely [the current documentation](http://martinos.org/mne/dev/manual/decoding.html).

Eric89GXL, 2017-03-06:
@choldgraf fixed all the links.

Are people working on any of these this week?

choldgraf, 2017-03-07:
I'm gonna take a pass through this and try to organize a little bit today...just been slow on doing it since I had to get the docathon up and running first :)  I think @teonbrooks is going to help out as well

choldgraf, 2017-03-10:
I'm looking through some of these issues etc now, and it's a little unclear to me what is the difference between "tutorials" and the "guide". Can somebody give me the high-level view of why these are two different things? Is the guide meant to be comprehensive and very MNE-specific, whereas tutorials are more conceptual in nature?

choldgraf, 2017-03-10:
E.g., what would be the difference between this page:

http://martinos.org/mne/stable/manual/time_frequency.html

and this page:

http://martinos.org/mne/stable/auto_tutorials/plot_sensors_time_frequency.html

agramfort, 2017-03-11:
hum... maybe http://martinos.org/mne/stable/manual/time_frequency.html should
not be there.

the "idea" is to have the manual more conceptual closer to the explanation
of the method
in a book or paper and to say when you should or should not use it. But
maybe it's too ambitious...

the tutorial are more really how to do things with MNE.

most of the manual content comes from the pdf doc of the MNE-C code.

choldgraf, 2017-03-11:
Maybe it'd be worth having language on the landing page that describes this? I remember when I was starting with MNE it was unclear which of the two I should be reading. I actually really like the idea of having more high-level concepts-based materials in the repo, since a lot of students learn by looking through project docs these days.

I think a challenge is that we'd need to be fairly vigilant about what belongs in tutorials vs. manual. For example, in tutorials there is [this general example on filtering](http://martinos.org/mne/stable/auto_tutorials/plot_background_filtering.html). If the manual is supposed to be more "concepts based" then this should really be in there. My assumption is that most users will get a feeling for what a section of the website covers by clicking a random link and seeing what kind of information is in there.

choldgraf, 2017-03-11:
And for a converse example, [this manual page on epochs etc](http://martinos.org/mne/stable/manual/cookbook.html#epoching-and-evoked-data) feels much more like a "how to use MNE" kind of document than a "general concepts" kind of document.

Maybe worth spending an hour at the meeting in NYC to chat about best way to organize / present this information? I think there's a lot of great info in there - wanna make sure that people will find it!

agramfort, 2017-03-11:
+ np.info['int64'].max

Alex

choldgraf, 2017-03-11:
sounds like all(opinion > 0 for opinion in thread_opinions) == True

I'll add to the project board

prateekj117, 2019-09-28:
@larsoner We should change the installation command for sphinx-autobuild and doc8 in the documentation from `conda install sphinx-autobuild doc8` to `conda install -c conda-forge sphinx-autobuild doc8` in case someone does not have conda-forge added to their channel list.

Rather than opening a separate issue, I think it would be more appropriate to add it here.

drammock, 2019-11-12:
@prateekj117 thanks for pointing that out, see #7050

drammock, 2020-11-19:
All of the remaining unchecked boxes in this todo-list have been added to #6794. Closing.

closed 2020-11-19 by drammock

choldgraf, 2020-11-19:
amazing!