mne-tools/mne-python #3987: ENH: plot_topo layout creation

Eric89GXL, 2017-02-14:
It seems like we keep hitting issues with topo scaling confusing people. I think we probably shouldn't change where (0., 0., 0.) is in any given channel layout, and instead ensure that our internal layouts are actually in head coordinates. I wonder if we could also just assume a head is e.g. 90 cm. That would make subselections of electrodes appear correct. A `normalize=True (default) | False` argument in functions that do such topologies could do it.

@jaeilepp WDYT?

jaeilepp, 2017-02-14:
Yeah, that sounds like a good idea. I was already experimenting with this, but it seems that ``_check_outlines`` may require some tweaking to make it work well with topomaps.

Eric89GXL, 2017-02-14:
This would fix #3814

jaeilepp, 2017-02-22:
I'm thinking that maybe the head coordinates should only apply when the layout is custom made. After all, the locations are only used in layout plots and they're not actual sensor locations. I noticed that vectorview locations for instance are not centered to 0. So some kind of adjustment would be needed anyway.
[image]

agramfort, 2017-02-22:
no opinion

Eric89GXL, 2017-02-22:
You mean the horizontal offset? I think this could actually be correct. If
you set dev_head_t to identity does it go away? If so it indicates that the
head was shifted to the side relative to the sensors, making this a feature
not a bug.

jaeilepp, 2017-02-22:
Why would it go away? These are just the sensor locations from the layout file.

Eric89GXL, 2017-02-22:
Then we should probably apply dev_head_t to get sensor locations in head
coordinates. Then the spatial relationship between the head and the
sensors, at least is terms of the center, will be correct.

jaeilepp, 2017-02-22:
But these are static locations from the layout file. Why would it help to transform them?

Eric89GXL, 2017-02-22:
If we are plotting sensors on top of a head, it would be best to plot the sensors with the same (or at 
least similar) geometrical relationship they actually had with the head during the recording. For MEG sensors this means using `dev_head_t` to get them to head coordinates.

Eric89GXL, 2017-02-22:
... so maybe not for `plot_topo` (and even then I'm not sure), but for `plot_topomap` it seems like we should always do things in head coordinates. Concretely the proposal would be:

1. Convert MEG sensors to head coords
2. Assume `(0, 0, 0)` is meaningful: don't re-center
3. Assume the head is about 90cm in diameter: don't rescale sensor positions
4. Any built-in EEG layouts that don't give "sensor locations in head coordinates" should be put through a function that effectively makes them be in head coordinates

This will give us much more realistic geometry than our current procedure, which is just to center based on extrema and scale everything to hit the boundaries.

jaeilepp, 2017-03-02:
> Any built-in EEG layouts that don't give "sensor locations in head coordinates" should be put through a function that effectively makes them be in head coordinates

Even some of our built-in MEG layouts don't give head coordinates (like KIT-AD). I'm not sure it's worth the effort to change this. After all, in most cases our layouts work fine and the change would require overhaul of all the plotting functions that use layouts. Also, topomaps use layouts so it might be dangerous to change this.

Eric89GXL, 2017-03-02:
I agree it would be a big change. Such a process should make all of the
plots more accurately represent the head to sensor relationship, though.
Actually we'd really want to skip the layout entirely and use the channel
positions from info. In the case of MEG sensors, we'd just need to apply
dev_head_t.

Maybe we could add it as a separate mode?

jaeilepp, 2017-03-02:
The layouts should still be used at least for topo-plots since the sensor triplets have identical positions.

Eric89GXL, 2017-03-02:
Ahh right. Forgot about that :)

And I guess there is some logic to the "unwrapping" transformation we normally do, since it's just one way (of many possibilities) to do a projection of spherical data.

Maybe just disabling norm and/or centering is enough

Eric89GXL, 2017-04-18:
@jaeilepp I think you have more or less fixed this, right?

jaeilepp, 2017-04-18:
Yeah, more or less. Maybe eventually it could be good to shift to convention you suggested, but I don't see a lot to be gained. Closing.

Eric89GXL, 2017-05-03:
This is essentially resurrected by #4239.

My proposal is that:

1. Any time we just want to select sensors (e.g., in `raw.plot`) we can use a layout with scaling, centering, etc.

2. Any time we want to show sensors in relationship to a head (topomap, joint plots, legends like in #4239) we should always put sensors in head coordinates. We should thus use real sensor positions, and not center or scale them. In the case of MEG sensors, this means using the `dev_head_t` and `chs[n]['loc']`. Otherwise we are mis-representing the relationship between the sensors and the head.

This will fix issues like `evoked.plot_topo()` and `evoked.pick_channels(evoked.ch_names(:10))` not having the 10 sensors in the same position (which is a bit crazy IMO).

Eric89GXL, 2017-05-03:
... so the distinction would be that Layout is only ever used for interactive "sensor selection" stuff, and never shown on a head. Only sensor positions in head coordinates ever get shown on a head (which could easily tie in eventually to the proposed Digitization class).

agramfort, 2017-05-03:
sounds reasonable to me !

larsoner, 2017-09-14:
This one won't happen for 0.15, moving to 0.16

larsoner, 2018-08-25:
Superceded by #5472 and #5471

larsoner, 2019-05-06:
Not fixed yet, so reopening

PR #7066 (2019-12-11): MRG, BUG: Fix M/EEG topomap plotting
Closes #3987
Closes #4880
Closes #5190 
Closes #5471
Closes #5472 
Closes #6304

Todo:
- [x] ensure our montages actually use the same head sizes (done by dig refactoring)
- [x] Make the `plot_sensors` be in physical coordinates
- [x] make all code assume a 0.095 m head radius
- [x] check that all examples in gist are fixed
- [x] deprecate params: `transform` in `montage`, `outlines`, ...
- [x] get tests to pass again
- [x] change to `sphere` rather than `head_radius` for consistency and flexibility
- [x] change units in `setup_volume_source_space` for cons [... cut]
closed 2019-12-11 by agramfort