mne-tools/mne-python #3990: Anyone used `custom_inherit` before?

choldgraf, 2017-02-14:
Somebody just pinged us about this on the docathon repo. It seems pretty handy for having one place for docstrings rather than having to re-type them everywhere:

https://github.com/meowklaski/custom_inherit

anyone used it or heard of it before?

Eric89GXL, 2017-02-16:
No. I think that SciPy uses `doccer` in `stats` to deduplicate. But I so far I haven't seen a case where we've really needed to deduplicate, beyond 1) the `*_doc_to_*_doc` functions that @wmvanvliet implemented, and 2) the `verbose` decorators that we have. There is some overhead both in implementation and in dev understanding to each new implementation / trick, so we would want to make sure it's worth it before trying.

choldgraf, 2017-02-16:
The main things I could see this being used for in MNE would be cases where, e.g., we have 2 functions doing the same thing, one for MNE objects and one for arrays. I agree it's not clear whether there's enough use for it though, just something to keep in mind. (if you want me to close this issue I can, just wanted to have some discussion on it)

Eric89GXL, 2017-02-16:
Yeah let's keep it in mind as we go, and reopen if someone has an idea where we should change something. In the meantime if anyone else has ideas feel free to add

closed 2017-02-16 by Eric89GXL