mne-tools/mne-python #3992: concatenate_epochs should check event_id values correspondence

mmagnuski, 2017-02-14:
I've recently had a weird bug - concatenating two epochs led to assertion error in epoch constructor. After some debugging I found out that the cause of the problem was that the epochs did not have the same event_id values (the same key corresponded to different values in these files). [This line in `concatenate_epochs`](https://github.com/mne-tools/mne-python/blob/master/mne/epochs.py#L2708) assumes that same keys correspond to same values and uses `update` method dict method to join dicts. It would be better if concatenate_epochs checked if this is true and threw error otherwise.

agramfort, 2017-02-15:
+1

can you open a PR?

mmagnuski, 2017-02-15:
sure, will open today

PR #3994 (2017-02-17): FIX: concatenate_epochs now checks for event_id correspondence
Fixes #3992

I'll add a note to whats new once CI's are green.
closed 2017-02-17 by agramfort