mne-tools/mne-python #3996: Question: maxwell_filter CPU usage?

jhouck, 2017-02-16:
I'm wondering how to predict/manage maxwell_filter's CPU hit.  Like maxfilter, it seems to follow the  OMP_NUM_THREADS environment variable.  Is there some more direct way to manage CPU for this or should I stick with os.setenv?

My CPU-starved coworkers thank you in advance for any advice.  ;)

Eric89GXL, 2017-02-16:
Can you change the var with e.g. `os.setenv` multiple times in a single session and have it take effect every time? If so we could cook up a `mkl_threads` decorator and add an `mkl_threads` argument to potentially heavy functions like `maxwell_filter`.

jhouck, 2017-02-16:
No, looks like once `maxwell_filter` runs the value sticks, even though os.getenv returns a new value each time it's changed.  That is, after an initial `os.environ['OMP_NUM_THREADS'] = '7'` , `maxwell_filter` will run with 7 cpus (even if OMP_NUM_THREADS was initially set to some other value in the shell), but subsequent changes have no effect on cpu use.  

In my case this will all be farmed out to the grid engine, so I can set the environment variable and -pe to the same value pretty easily.  Maybe not ideal otherwise.  Is OpenMP's influence sneaking in via linalg or something?

agramfort, 2017-02-16:
yes OpenMP is used in linalg via the MKL library. We don't do anything
special

Eric89GXL, 2017-02-16:
@jhouck do you have time to make a PR to the FAQ with something like "A function uses multiple CPU cores even though I didn't tell it to, why?" with a few sentences: MKL using OpenMP, the OMP_NUM_THREADS option, and a line about how to set it (with env var before running python like `OMP_NUM_THREADS=1 python ...` or with `import os; os.setenv(...)` before running any functions)?

jhouck, 2017-02-16:
Sure, I could do that.  Probably not until the weekend though.

closed 2017-02-24 by jhouck