mne-tools/mne-python #3997: Meaning of l_freq/h_freq in filter functions

cbrnr, 2017-02-16:
I was wondering which values are expected by `l_freq` or `h_freq` in `filter.filter_data` (or `create_filter`). The docstring says cutoff frequencies, but the cartoons in the notes section below indicate that it could also be edge frequencies. Even if they are cutoff frequencies, it is still unclear which definition is used - -3dB or -6dB? Usually, -3dB cutoff frequencies are reported for IIR filters and -6dB for FIR filters.

Eric89GXL, 2017-02-16:
Our IIR filters are forward-backward, so I think the frequencies are -6dB points.

Regarding FIR -- the `l_freq` and `h_freq` are the places we pass `1`'s in the FIR filter design functions, and `l_freq - l_trans_bandwidth` and `h_freq + h_trans_bandwidth` are where we put the `0`'s. I'm not sure offhand whether or not this produces e.g. 0dB, -3 dB, or something else at those points because the `firwin2` windowed design method does some smoothing.

cbrnr, 2017-02-16:
Thanks. How would you report the filter specs then? It seems like knowing what these frequencies correspond to is an important point. I'm not that familiar with filter design (but will dig in some more), but I assume that there must be some common parameters that can be reported for any filter type (independent of the design method and FIR/IIR)?

Eric89GXL, 2017-02-16:
Depends on how specific you need/want to be. Usually I have been lazy and just given e.g. `h_freq` as the "lowpass frequency", because my analyses haven't critically depended on these parameters. But if you are doing something like tight band-passing, then it will probably matter and we'll have to dig in deeper.

closed 2017-02-16 by cbrnr