mne-tools/mne-python #3998: EEG (re-)referencing is confusing

hoechenberger, 2017-02-16:
Hello, we were struggling with the issue of (re-)referencing EEG data in MNE, and I would like to share with you what caused our trouble, and suggest and discuss potential solutions.

Loosely following [the re-referencing example](http://martinos.org/mne/dev/auto_examples/preprocessing/plot_rereference_eeg.html), we did:

```python
import numpy as np
import matplotlib
import os
import mne
from mne.datasets import sample

matplotlib.use('Qt5Agg')

data_path = sample.data_path()
raw_fname = os.path.join(data_path, '/MEG/sample/sample_audvis_filt-0-40_raw.fif')
event_fname = os.path.join(data_path, '/MEG/sample/sample_audvis_filt-0-40_raw-eve.fif')
event_id, tmin, tmax = 1, -0.2, 0.5

raw = mne.io.read_raw_fif(raw_fname, preload=True)
events = mne.read_events(event_fname)

picks = mne.pick_types(raw.info, meg=False, eeg=True, eog=True, exclude='bads')
reject = dict(eeg=180e-6, eog=150e-6)
epochs_params = dict(events=events, event_id=event_id, tmin=tmin, tmax=tmax,
                     picks=picks, reject=reject, add_eeg_ref=False, preload=True)

# Let's remove all existing projectors from the example data set,
# since in "real life", we start with just the raw data.
raw.del_proj()

epochs = mne.Epochs(raw, **epochs_params)
```
During data cleaning, we'd sometimes switch between different reference electrodes.

```
epochs.set_eeg_reference(['EEG 001']).plot()
epochs.set_eeg_reference(['EEG 010']).plot()
```
After the data is cleaned, we'd like to apply an average reference and view the evoked potentials. Thanks to #3756, one can easily switch to an average EEG reference, even if another "custom" reference had been applied earlier. 
```python
evoked = epochs.set_eeg_reference().average()
evoked.plot()
```
What we expected to look like a butterfly plot actually looked more like a butterfly that barely survived a freak accident. We were at a loss as to what went wrong. Later, I looked a bit closer at the output produced by the first line:
```python
Adding average EEG reference projection.
1 projection items deactivated
```
Until then, we had simply managed to entirely ignore these messages:
*Projection? Never heard of it. Must be an MEG thing that doesn't concern us!*

However, doing the following instead actually did do the trick:
```python
evoked = epochs.set_eeg_reference().apply_proj().average()
evoked.plot()
```

Neat! Maybe let's swtich back to another custom reference just to check something:
```python
epochs.set_eeg_reference(['EEG 002'])
```
Let's check out `epochs.info` for the sake of it (I'm showing only the relevant lines here):
```
In[65]: epochs.info
Out[65]: 

<Info | 19 non-empty fields
    custom_ref_applied : 'bool | True
    projs : 'list | Average EEG reference: on
>
```
So now we somehow have both, an average reference (applied projection) and our custom reference (`EEG 002`). The projection cannot be removed, so now I'm stuck forever and always with the average reference.

But wait, there might be a workaround!

```python
epochs = mne.Epochs(raw, **epochs_params)
evoked = epochs.set_eeg_reference(epochs.info['ch_names']).average()
evoked.plot()
```
Looks good!

Let's take a peek at `epochs.info` again:
```python
In[69]: epochs.info
Out[69]: 

<Info | 19 non-empty fields
    custom_ref_applied : 'bool | True
    projs : 'list | 0 items
>
```
Although we effectively use an average reference, apparently no projector was added this time.

We can now also easily switch back and forth between average and custom references:
```python
In[83]: np.allclose((epochs
   ...:              .copy()
   ...:              .set_eeg_reference(['EEG 002'])
   ...:              .get_data()),
   ...:             (epochs
   ...:              .copy()
   ...:              .set_eeg_reference(epochs.info['ch_names'])
   ...:              .set_eeg_reference(['EEG 002'])
   ...:              .get_data()))
   ...: 
Applying a custom EEG reference.
Applying a custom EEG reference.
Applying a custom EEG reference.
Out[83]: 
True
```
This works exactly as I would have expected.

I find this behavior somewhat confusing, because it appears kind of inconsistent to me: sometimes, `Epochs.set_eeg_reference()` will add what I would commonly consider an *ordninary* EEG reference; but on some occasions, it it would create this weird thing called a *projector* that takes an additional step to activate, and then it's forever stuck to the data. This is nothing I've ever seen in any EEG papers I've read so far.

In my opinion, `Epochs.set_eeg_reference()` should never add a projector unless explicitly requested. `Epochs.set_eeg_reference(ref_channels=None)` should effectively translate into `Epochs.set_eeg_reference(epochs.info['ch_names'])` instead of adding a projector. 

In the  [example code](http://martinos.org/mne/dev/auto_examples/preprocessing/plot_rereference_eeg.html) I mentioned earlier, the (average reference) projectors are implicitly automatically applied when constructing the `Epochs` objects (since `proj=True` by default in `Epochs.__init__()`). For that
reason, we entirely overlooked the need to call `apply_proj()`: In our workflow, average references
are added to an existing `Epochs` instance once all the data cleaning has been done, immediately before calculating GFP etc. However, `Epochs.average()` does not apply the projectors automatically, so we ended up with weirdly looking "butterfly" plots.

I've asked several researchers who'd been working with EEG data for several years now whether they could spontaneously associate the words *EEG*, *projector*, and *projection* in the context of EEG pre-processing and potentially artifact removal. The responses were all negative.

Even if you believe I might be kind of ignorant or stupid by now, I do believe that either small API or implementation changes (as proposed here), enhanced documentation, and "improved" / new example code could make MNE much more approachable to those who have no experience with MEG so far. I'd be willing to help working on this, but first I will have to finish my work on #3967, which has been stalled for a while now already...

Eric89GXL, 2017-02-16:
> What we expected to look like a butterfly plot actually looked more like a butterfly that barely survived a freak accident.

:)

> In my opinion, Epochs.set_eeg_reference() should never add a projector unless explicitly requested. 
> ... I find this behavior somewhat confusing, because it appears kind of inconsistent to me

It is indeed inconsistent. From a source imaging (and historical) perspective, projections are preferred, so we use them whenever possible. So for our purposes let's assume we need to stick with it, and figure out how to best make the situation better for EEG folks.

Can you think of a good way to modify the docstring for `set_eeg_reference` that makes this approach clearer? It would be especially useful if you could open a PR to add better text to that function. We should also change [the reference example](https://mne-tools.github.io/stable/auto_examples/preprocessing/plot_rereference_eeg.html) into a tutorial at some point. Feel free to do that if you want, but even without doing that, can you look to see if there are things there that could be made clearer?

> projectors are implicitly automatically applied when constructing the Epochs objects

This is something that has come up a few times as being problematic, actually: #3664, #3410, #2310... still trying to figure out the best solution.

> Until then, we had simply managed to entirely ignore these messages:
> Projection? Never heard of it. Must be an MEG thing that doesn't concern us!

How about rephrasing to:
```
Adding average EEG reference as a projection operator.
Use apply_proj() to apply the average projection.
```
Would that have worked? If so, I can add that.

> So now we somehow have both, an average reference (applied projection) and our custom reference (EEG 002). The projection cannot be removed, so now I'm stuck forever and always with the average reference.

That's a bug. @wmvanvliet do you have time to try to fix it? The projector should be removed if a new reference is set.

> Even if you assume I might be kind of ignorant or stupid by now

I [agree with Don Norman](https://www.amazon.com/gp/product/0465050654) that things like this should not be so hard, and that user struggles usually due to API and documentation. In this case I think the API is pretty settled (beyond the bug mentioned above), but hopefully we can make the docs better.

>  first I will have to finish #3967, which has been stalled for a while already...

Looking at that issue, I think it needs a couple of (hopefully) small changes. So feel free to tackle that one first if you want. But most contributors are guilty of having multiple PRs open at once, so don't let #3967 stop you from tackling this, especially since the considerations here are fresh in your mind.

hoechenberger, 2017-02-17:
Hi Eric, thanks for your response!

> It is indeed inconsistent. From a source imaging (and historical) perspective, projections are preferred, so we use them whenever possible.

Ah, I already suspected that! What weird guys you are! ;) (Just kidding, of course. Very interesting how approaches can differ!)

> So for our purposes let's assume we need to stick with it, and figure out how to best make the situation better for EEG folks.
>
> Can you think of a good way to modify the docstring for [...]

I do have some ideas on my mind, but I think a (backward-compatible) API change could be a better approach. I will explain below.

> How about rephrasing to:
> 
> Adding average EEG reference as a projection operator.
> Use apply_proj() to apply the average projection.
> Would that have worked? If so, I can add that.

This would be a more clear message indeed, yes. Still I wonder if it would actually have helped many EEG folks: I believe the major issue is non-familiarity with a specific concept (projections) instead of actual phrasing. **Remember, I actually somehow managed to completely ignore the whole projector thing at first!** So we're dealing with a large amount of persistent ignorance here; this is not to be taken lightly! No but seriously: assuming we insisted on always doing average referencing via a projection no matter what, the message should be even more extensive, something along the lines of:
```
Adding average EEG reference as an INACTIVE projection operator.
!!! The EEG data is not actually referenced to average yet !!!
You must call apply_proj() to apply the projection and reference to average.
```
... and add some blinking exclamation marks and such ;)

> So now we somehow have both, an average reference (applied projection) and our custom reference (EEG 002). The projection cannot be removed, so now I'm stuck forever and always with the average [...]

I was referring to a different issue here :) Namely that you cannot reverse a projection operation:

```python
(epochs
 .copy()
 .set_eeg_reference()
 .apply_proj()
 .del_proj())
Adding average EEG reference projection.
1 projection items deactivated
Created an SSP operator (subspace dimension = 1)
1 projection items activated
SSP projectors applied...
Traceback (most recent call last):
  File "/Users/hoechenberger/anaconda3/lib/python3.5/site-packages/IPython/core/interactiveshell.py", line 2881, in run_code
    exec(code_obj, self.user_global_ns, self.user_ns)
  File "<ipython-input-7-95a9933f4fb3>", line 1, in <module>
    (epochs
  File "/Users/hoechenberger/Development/mne-python/mne/io/proj.py", line 200, in del_proj
    raise ValueError('Cannot remove projectors that have already '
ValueError: Cannot remove projectors that have already been applied
```
Which brings me back to idea regarding a potential API change.

Currently, `set_eeg_reference()` accepts its argument `ref_channels` to be `None`, an empty list (which will probably be removed, according to #3981), and a list of channel names (actually, it also accepts strings and then tries to treat them like lists, which usually won't work... but that's another issue :D).

How about we additionally allow `ref_channels='average_proj'` (which would behave like `ref_channels=None`) and `ref_channels='average'`. The latter would *not* create a projection, but do something like
```python
set_eeg_reference(epochs.info['ch_names'][first_eeg_chan:last_eeg_chan+1])
```
as I demonstrated in my earlier posting. (Please note that there was a bug in my original posting: it would not exclude the EOG channel. This is corrected now. BTW is there any easy way to figure out which channels belong to MEG, EEG, EOG etc. without parsing `info['ch_names']`?)

This approach
- would remove the need of having to apply the average projector and
- therefore allow users to switch back and forth between "custom" and average references as desired,
- thereby making the behavior much more consistent
- would make the API very explicit for both EEG and MEG users
- would make the API more approachable for EEG folks and avoid confusion, as they could simply entirely ignore the projector parts.

What do you think?

> first I will have to finish #3967, which has been stalled for a while already...
> Looking at that issue, I think it needs a couple of (hopefully) small changes. 

Yes, my first approach wasn't tackling the actual problem, I assume. But I think I've figured it out in the meantime; just didn't have the time to implement the changes :)

Thanks for your support!

hoechenberger, 2017-02-17:
> I agree with Don Norman that things like this should not be so hard, and that user struggles usually due to API and documentation. In this case I think the API is pretty settled (beyond the bug mentioned above), but hopefully we can make the docs better.

Wow, thanks for the reference! I just ordered this book.

agramfort, 2017-02-19:
can someone take care of the bug fix reported here? thx

wmvanvliet, 2017-02-20:
I'll take a look

wmvanvliet, 2017-02-20:
I verified that the following works as expected:

```python
epochs.set_eeg_reference().apply_proj()  # Set average reference
epochs.set_eeg_reference(['EEG 001'])  # Re-reference to channel EEG 001
```

There is no bug here. When you look at the info structure, you indeed see this:

```python
In[65]: epochs.info
Out[65]: 

<Info | 19 non-empty fields
    custom_ref_applied : 'bool | True
    projs : 'list | Average EEG reference: on
>
```

An "average" projector that has been applied and a custom reference that has been applied. Projectors are never removed from the info structure. When they have been applied to the data, they are flagged as such and thereafter become inert. I agree that the term "active" and it's value "on" are confusing. It conveys the message that the projection is actively being applied (hence overriding the custom reference) but it's not, it indicates that the projection has been applied somewhere in the past.

We're not likely to change the behavior of the projection code. We can tweak the printing of `info` a bit if we can agree on less confusing terminology, but most of all, this is a documentation issue. 
@hoechenberger 's initial code is completely right, does exactly what he hoped to achieve, but still managed to confuse him :) I agree with @Eric89GXL that it would be good to expand the referencing example in a comprehensive tutorial.

wmvanvliet, 2017-02-20:
the behavior of projectors is actually more complicated then I led on. When `Epochs` are created with `preload=False`, an active projector is indeed actively being applied when data is being accessed. There may actually be a bug here. When doing `epochs.set_eeg_reference()`, and the data is preloaded, the average ref projector should be applied straight away, because the `Epochs` object will not do so automatically in `preload=False` mode.

You know what? I'll take a shot at writing a tutorial on the behavior of projectors first. They still confuse me even after reading through the code 5+ times.

agramfort, 2017-02-20:
well when changing the EEG ref manually you undo the EEG Average Ref so
technically the proj is not On anymore. I think the EEG average ref shoud
be marked as off as soon as you undo it.

wmvanvliet, 2017-02-20:
that would cause the average reference to be re-applied if you ever called `.apply_proj()` again

wmvanvliet, 2017-02-20:
to make it actually be gone, it should be completely removed from the projections list.

agramfort, 2017-02-20:
>
> that would cause the average reference to be re-applied if you ever called
> .apply_proj() again
>
yes but is it a problem?

wmvanvliet, 2017-02-20:
yes, because some functions (such as plotting evokeds) will call `inst.apply_proj`, expecting it to be a null operation of projections have already been applied.

wmvanvliet, 2017-02-20:
or at least expecting that the user would like all the projections to be applied. The code currently cannot deal with projections that the user *doesn't* want to be applied.

Eric89GXL, 2017-02-20:
It should be okay if we also remove it from the list

agramfort, 2017-02-20:
sounds like a plan

nbara, 2017-06-26:
@wmvanvliet I'm a bit puzzled as to what `inst.apply_proj()` actually does. 

Here is a gist where I first set `ref_channels=[]` (as my data was already referenced).

https://gist.github.com/nbara/972eefa4a62fd6ef9df308b76ffa2559

Applying apply_proj() not only rereferences to the mean, but also appears to include the STIM channel ??

mmagnuski, 2017-06-26:
Another reref issue: https://github.com/mne-tools/mne-python/issues/4352, we should probably open this one as rereferencing is not intuitive enough - at least until the agreement reached in https://github.com/mne-tools/mne-python/issues/4302 is implemented.

Eric89GXL, 2017-06-26:
Agreed, might as well reopen to consider during #4302's implementation

Eric89GXL, 2017-07-20:
Closed by #4382. @hoechenberger feel free to try latest `master` to see if it makes more sense now

closed 2017-07-20 by Eric89GXL

hoechenberger, 2017-07-27:
@Eric89GXL This is fine, thanks for following up!

PR #4004 (2017-02-22): [MRG+2] Make EEG (re-)referencing more intuitive
To make (re-)referencing of the EEG a bit more intuitive, this PR changes the following:

 - Moving from an average to a custom reference completely removes the average reference projection from the `info['projs']` list. A message is printed when this happens.
 - When setting an average reference on data that has been preloaded, the projection is automatically applied.
 - The referencing example has been updated to re-reference the same data, instead of making copies of the original un-referenced data. This emphasizes that it is possible to free [... cut]