mne-tools/mne-python #4001: Examples for using MNE-Python directly in Matlab

ftadel, 2017-02-17:
Hello

I spent the last few days exploring the support for Python in Matlab 2016b. I was very surprised, it's working beautifully! Matlab can be used with little reformatting as an interactive Python console, and MNE-Python objects and functions can be used directly from Matlab scripts.
This could be very interesting for a lot of people in the community (and for me too: I'm hoping to be able to make some of your recent developments available to the Brainstorm users)

As an example, I wrote a Matlab version of plot_brainstorm_data.py: [plot_brainstorm_data.zip](https://github.com/mne-tools/mne-python/files/784060/plot_brainstorm_data.zip)

I will distribute it on my end as part of the Brainstorm distribution, but I think it would make sense to include it in the mne-python examples too. I was suggesting to add this script to mne-python/examples/datasets, because the link between the python and matlab versions of the same thing would be very clear. But Alex is concerned by the fact that maintaining it will be complicated.

Any suggestion?

Thanks
Francois

agramfort, 2017-02-18:
@ftadel can you make your example script available via https://gist.github.com/ so we can easily see the content? thx

dengemann, 2017-02-18:
Amazing news! I am very excited. Please keep us posted.

ftadel, 2017-02-18:
@agramfort https://gist.github.com/ftadel/ec388c078e9b78bfe97eb705be1f7b7f

dengemann, 2017-02-18:
Looks surprisingly clean. It seems you can indeed easily create matlab
wrappers around some functions we so far only have in Python. Really good
news! Thanks for sharing Francois.

jasmainak, 2017-02-18:
This is great! I've heard many users complain about interoperability between packages. The gist is a great starting point

kingjr, 2017-02-18:
Waow.

(It's going to be even harder to convince people to quit Matlab, but still
very cool)

Eric89GXL, 2017-02-20:
That's awesome!

> I was suggesting to add this script to mne-python/examples/datasets, because the link between the python and matlab versions of the same thing would be very clear. But Alex is concerned by the fact that maintaining it will be complicated.

Hmmm... if we put it there, it won't render properly in our automated CI doc builders because they don't have MATLAB. So that poses some challenge.

If the MATLAB calling requirements change, then we'd need to change it. At the same time, if we change something in MNE that improves the example, we'd also need to improve the MATLAB-ized version of that example. So this is also some overhead.

But I agree it would be nice to have it somewhere visible.

One option would be to add an "interoperability" section somewhere in the docs where we could point out this feature?

agramfort, 2017-02-20:
I am also concerned about the continuous integration of such a code.

@ftadel if you have this on the brainstorm website would it be ok to
add a section on the MNE doc and point to the brainstorm page to see
the code?

ftadel, 2017-02-24:
I will document how to connect Brainstorm and MNE-python when I have some interesting features to offer at a higher level (like running your implementation of SSS from Brainstorm). 

At a lower level, I thought you guys might be interested in explaining to your users how to call MNE-python from Matlab scripts, so I shared this example. Feel free to do whatever you want with it.

larsoner, 2019-08-20:
Closing but let's keep this stuff in mind !

closed 2019-08-20 by larsoner