mne-tools/mne-python #4012: ENH: Better evoked.comment from epochs[cond].average()

Eric89GXL, 2017-02-23:
From the mailing list, people get `epochs[cond].average().comment == 'Unknown'`.

Deep in our code it uses `epochs.name`, which I didn't even know existed, but defaults to `Unknown`. We could set this name to something at least when doing `epochs[cond]`, which would then propagate to `epochs[cond].average()`.

jaeilepp, 2017-02-23:
Also, the ``name`` attribute is not saved or read.

Eric89GXL, 2017-02-23:
From `git grep` it looks like `epochs.name` is almost unused. And if it's not read/written it's not very useful. I'm +1 for getting rid of it.

jaeilepp, 2017-02-24:
> I'm +1 for getting rid of it.

Me too.

jaeilepp, 2017-03-02:
On further investigation, it seems that it tries to read it, but the save function does not save it. There is also a FIFF constant for it.

jaeilepp, 2017-03-02:
So I'm wondering if we should stick to the old way, but just make sure it gets saved?

Eric89GXL, 2017-03-02:
I can live with that. So we fix the save, and fix the comment creation.

closed 2017-03-22 by Eric89GXL