mne-tools/mne-python #4016: raw.resample drops events

timmidee, 2017-02-24:
When I resample data from 1000Hz to 100Hz, some events are dropped.

I go from 10690 to 10645 events before and after resampling, as found by mne.find_events(raw,shortest_event=0,consecutive = True).

Is this expected behaviour?

I realise I have an awful lot of events, but that's because I have eye movement events coded as events.
It also means events are sometimes on adjacent samples (hence consecutive = True). Could it be that the resampling causes temporally (almost) adjacent events to end up on exactly the same sample and therefore drops (one of) them? (cf. #3938)

Eric89GXL, 2017-02-24:
Yes this can happen. What you can do if you want to preserve them (but note you will lose timing precision!) is do `events = find_events(raw_orig, ...)`, and then rescale the event times to the new sample rate with something like:
```
events[:, 0] = np.round(events[:, 0] * (new_rate / float(old_rate)).astype(int)
```
Does that make sense?

Can you think of a good place to put this sort of thing in the docs? `raw.resample`? The FAQ?

timmidee, 2017-02-25:
Makes sense, thanks! 
And in case you were asking me directly: I think it might be good to have with the help for raw.resample. The explicit warning for jitter of the events and epoching is already good to have there, adding a clear sentence or two about the possible consequences for temporally adjacent/near events would fit right in with that IMO.

Eric89GXL, 2017-02-25:
Do you have time to make a quick PR?

timmidee, 2017-03-01:
Sorry, I'm a bit of a noob when it comes to the larger scheme of developing software (github, docs, etc) what's a PR?

jaeilepp, 2017-03-01:
PR is short for pull request (github lingo). See https://mne-tools.github.io/stable/contributing.html?highlight=contributing to get started.

jona-sassenhagen, 2017-03-28:
So this is different from the problem fixed in https://github.com/mne-tools/mne-python/pull/2298 ?
`raw, events = raw.resample(srate, events=events)`

timmidee, 2017-03-28:
If I read correctly, then #2298 deals with events ending up on adjecent samples, e.g. when downsampling 0 0 1 0 1 0 0 0 yields 0 1 1 0 (rendering it to only one value-change on the stim  channel). This issue is more about events ending up on the exact same sample, which is "impossible" on one single stim channel with one value per timepoint.

timmidee, 2017-03-29:
Sorry, my fingers must have slipped on my phone, not sure this is resolved. Didn't mean to close it. Seems different from #2298.

Eric89GXL, 2017-04-18:
@timmidee are you ready to make a PR now? :)

timmidee, 2017-04-24:
I think I need a solid python editor, a crash course in GitHub/forking/PR/upstreams/trunks/branches/etc. and more time than my PhD currently allows... Maybe I can write a few sentences and bug one of you to make the PR for me while I slowly learn this stuff?

Eric89GXL, 2017-04-25:
Sure. Or take the (hopefully) easy route and use the GitHub interface to make a PR.

agramfort, 2017-07-22:
feel free to reopen if necessary

closed 2017-07-22 by agramfort