mne-tools/mne-python #4021: ENH/DOC cHPI support for artemis123 data

bloyl, 2017-02-27:
Hey @Eric89GXL 
I'm interested in evaluating the cHPI code for localizing HPI coils in artemis123 (and CTF) but the code makes use of header information that I don't know how to recreate/format.

Overall, is there any interest in refactoring this code to be less elekta centric. I'd be willing to take a stab at it but would need some guidance on the existing code.

For instance just to get started understanding the code. I would need some information about the following fif fields that end up in the info structure.

1) hpi_meas : Seems like 'number' and 'coil_freq' are the used parameters here.
2) hpi_subsystem : seems like https://github.com/mne-tools/mne-python/blob/master/mne/chpi.py#L191-L202 is the relevant section but its unclear on what the role of hpi_pick is later on.
3) hpi_results : Seems to be the results of the elekta HPI fitting done in prescan. This is mostly used as a check but also to do some transformation of the digitization points to bring them into device cords maybe?

Tangentially related question, what is the status of the Dig class refactoring? I need methods for reading polhemous (.pos) files similar to https://github.com/mne-tools/mne-python/blob/master/mne/io/ctf/eeg.py#L56-L57 and don't want to duplicate code if this is already being supported more generally?

Thanks
Luke

Eric89GXL, 2017-02-27:
> Overall, is there any interest in refactoring this code to be less elekta centric. I'd be willing to take a stab at it but would need some guidance on the existing code.

I think about it the other way -- let's get our non-FIF file reader(s) to populate `info` properly, as opposed to making the cHPI code itself use other formats. That way things like saving, etc. will work. The nice thing about the Elekta Neuromag FIF format is it contains (almost) everything we would want, many parts of our code make use of it, it's our standard IO format, etc.

You've actually already done most of the work by enumerating which fields you need to populate properly, so I think you're really headed that way anyway...

> its unclear on what the role of hpi_pick is later on.

It's basically a STIM channel that says which cHPI coils were active. It's not 100% necessary, though. If the code currently throws an error without it, we can fix that.

> This is mostly used as a check but also to do some transformation of the digitization points to bring them into device cords maybe?

Yeah. The coil positions in head coordinates might also be used, I'd need to check. IIRC from this part of the code, we need coil coordinates in the head frame, plus starting coil positions in the device frame. Having `info['dig']` setup properly and having `info['dev_head_t']` should thus provide a suitable fallback mechanism.

> I need methods for reading polhemous (.pos) files similar to ...

Argh, I didn't realize that was just Polhemus data reading when I translated Matti's code. Can you see if `read_dig_montage` works for you? That is the general digitization reader. We should refactor the CTF code to use it. And when we eventually refactor digitization stuff, `read_dig_montage` will stay.

bloyl, 2017-02-27:
Sorry, I should have been more clear on the refactoring suggestion. I am not proposing that we use non-fif fields in info to drive the cHPI fitting. but rather to isolate the key pieces of info the are needed to proceed from the rest.

At the moment my thought is to refactor _get_hpi_info to remove the requirement for 'HPI_results' and 'dig'. The rational is two fold:
1) They aren't really needed to do continuous HPI localization in device coords
2) For artemis data I don't have 'HPI_results' because it isn't done on the scanner and for a number of subjects I want to be able to localize the HPIs in device coordinates even without a digitization.

I'll look at read_dig_montage I imagine that is what i need.

Eric89GXL, 2017-02-27:
> They aren't really needed to do continuous HPI localization in device coords

True -- if all you want is a time-varying estimate of the location of each coil, you don't need anything other than frequencies of interest, really. But keep in mind that for the vast majority (99%?) of users, the point of cHPI fitting is to get something more complete -- a reliable, time-varying `dev_head_t` estimate.

The routines for estimating this transform currently rely on having some "ground truth" for where the electrodes are in relation to one another (and also relies on, but less strongly so, on having some initial position for each one in the device). Doing so has some potential benefits in the context of estimating `dev_head_t`, like hopefully making the fitting recognize a faulty electrode position estimate so it can be discarded. (Although I've been thinking lately that a more general fitting routine, which could possible update this ground truth to e.g., reflect slow changes in electrode position, could work even better.)

In any case, that's just to say that we'll have to be careful about splitting functions in a way that doesn't hamper the main use case, yet allows you to get access to individual coil fits in some sensible way. I'll let you think about how to do it :)

bloyl, 2017-03-07:
So the refactoring was/is pretty straight forward. the key test (I think) test_calculate_chpi_positions should keep passing.  Other tests, that depend on the order of the logging messages, might need to be adjusted/changed but we can talk about that later.

I am now stuck dealing with digitization points and the various coordinate systems and their associated transforms.

So here are my questions.
1) What is the most supported head coordinate frame through out MNE? What choice is least likely to cause me problems in the future with this data.
        I assume that it is FIFF.FIFFV_COORD_HEAD           = 4.

2) Does the description of the neuromag coord frame from  [here](http://www.fieldtriptoolbox.org/faq/how_are_the_different_head_and_mri_coordinate_systems_defined) correctly describe it?

3) Are there utilities for mapping between the established head frames? For instance can I transform a  dig structure from FIFFV_MNE_COORD_CTF_HEAD to  FIFFV_COORD_HEAD?

4) Can I mix coord systems within the same structure? can I have some Dig points in COORD_HEAD and the HPI points in COORD_DEV?

thanks

Eric89GXL, 2017-03-07:
> What is the most supported head coordinate frame through out MNE?

Yeah you want them in Neuromag head coords (FIFFV_COORD_HEAD).

> Does the description of the neuromag coord frame from here correctly describe it?

Yes

> Are there utilities for mapping between the established head frames?

Given LPA/Nasion/RPA (in any coordinate frame) you can get the transformation to Neuromag head coordinates. I can't remember the function offhand, but it's used in `read_dig_montage`. Hopefully that's all you need?

> Can I mix coord systems within the same structure? can I have some Dig points in COORD_HEAD and the HPI points in COORD_DEV?

`info['dig']` supports this, yes. But having HPI points in device coordinates will not help with head position fitting -- you need the "digitized in head coordinates" versions.

bloyl, 2017-03-07:
read_dig_montage doesn't support these files. The format is different from the elp files that I guess fastscan exports. These 'pos' files are generated by program written by CTF that uses the polhemous isotrack systems. they are pretty straight forawrd and supporting them would be easy. Let me know if you'd like me to add it to read_dig_montage. There is also the brainstorm polhemous files (http://neuroimage.usc.edu/brainstorm/CoordinateSystems#Brainstorm_.pos_files) which could be supported at some point.

While info['dig'] seems to support it( each point has its own COORD_FRAME). its not clear that this info is used. for instance plot_trans in viz._3d.py looks like it assumes all dig points are in head space. It isn't a big deal either way I was just curious.

Eric89GXL, 2017-03-08:
Feel free to add support to `read_dig_montage` for another format. For now let's just add the file format(s) you actually need / use and assume YAGNI until we hit a use case for others (e.g. Brainstorm).

> While info['dig'] seems to support it( each point has its own COORD_FRAME). its not clear that this info is used.

Yes this is buggy behavior not to check the coord frame. We need to fix it, and I'll probably try to do some of it as part of #3893.

bloyl, 2017-03-14:
I decided not to add support to read_dig_montage. It would have required multiple calls to read_dig_montage to parse the different point types plus my own pass of the file to do sorting based on labels. I also noticed lots of overlap/interconnections between read_dig_montage, and _read_dig_points and _make_dig_points not sure if that's on your agenda for #3893 but thought i'd bring it up.

I just did a PR #4069 for a code review. tests are passing except for some docstring thing. that I'm not sure how to fix to be honest. Before I proceed to far I'd like your opinion on the refactoring of chpi.py. Particularly the splitting up of _get_hpi_info?

larsoner, 2019-08-20:
Fixed by #4069

closed 2019-08-20 by larsoner