mne-tools/mne-python #4026: BUG: test magnetic dipole

SherazKhan, 2017-03-01:
@Eric89GXL 
Hi Eric,

I was working on real time chpi, stumble upon `test_magnetic_dipole()` in test_make_forward.py :-)

Input `r0 = np.array([0., 13., -6.])` is in mm however coils locations are in meters `_magnetic_dipole_field_vec(rrs, coils):` should not we divide by 1000: `r0 = np.array([0., 0.13., -0.06.])/1000.`

may be a PR like this
`def test_magnetic_dipole():
    """Test basic magnetic dipole forward calculation."""
    trans = Transform('mri', 'head')
    info = read_info(fname_raw)
    picks = pick_types(info, meg=True, eeg=False, exclude=[])
    info = pick_info(info, picks[:12])
    coils = _create_meg_coils(info['chs'], 'normal', trans)
    # magnetic dipole at device origin
    r0 = np.array([0., 13., -6.])/1000.
    for ch, coil in zip(info['chs'], coils):
        rr = (ch['loc'][:3] + r0) / 2.
        far_fwd = _magnetic_dipole_field_vec(r0[np.newaxis, :], [coil])
        near_fwd = _magnetic_dipole_field_vec(rr[np.newaxis, :], [coil])
        ratio = 8. if ch['ch_name'][-1] == '1' else 16.  # grad vs mag
        assert_allclose(np.median(near_fwd / far_fwd), ratio, atol=1e-1)`

Eric89GXL, 2017-03-01:
Hah! Yeah, sure. Although this makes me think that it's actually okay/useful to test both a dipole that is at device origin (your idea and what I meant to do) and also a dipole that is far from the array (what I actually implemented).

PR welcome :)

SherazKhan, 2017-03-01:
Thanks @Eric89GXL 
Will prepare the PR with both test cases.

I am currently testing your chpi code on large dataset and comparing with maxfilter. You did an awesome job :-)

Eric89GXL, 2017-03-02:
I had lots of help from the Elekta folks to get it going. @bmaess most recently fixed some bugs, too. Hopefully you can find ways to make it better, too

Eric89GXL, 2017-04-18:
Ping @SherazKhan do you have time for this, or should I take over?

Eric89GXL, 2017-08-05:
Fixed in https://github.com/mne-tools/mne-python/commit/6a03bed7cb5c8147d645010ab404c376bf1d885b

closed 2017-08-05 by Eric89GXL