mne-tools/mne-python #4027: Apply ICA after autoreject

ramonapariciog, 2017-03-01:
Hi everybody,
I'm working to get an ERP (Mismatch Negativity) with records with EGI NetStation and montage 'GSN-HydroCel-129'.
The data has a lot of artifacts and i apply the autoreject tool for clean it but the result stills noisy.
I'm applying butterworth high (0.5) and low (40 Hz) pass filter order 5, separately, and notch (50 and 100 Hz).
I'm trying to get an eog channel from the frontal eeg channels, because the data doesn't have eog channel originally, then i add the new channel to the raw, i get the epochs, i apply autoreject and finally, i want to apply ICA decomposition over the cleaned epochs. My question is: Autoreject can work with the data that includes the eog channel or how can i addapt the eog_index obtained by ica.find_bads_eog and create_eog_epochs, if autoreject remove bad epochs and channels automatically.
This is my code:

`mne.io.set_bipolar_reference(raw, ['EEG 126'], ['EEG 127'],
                             ch_name=['EOG'],
                             ch_info=dict(ch_types='eog'),
                             copy=False)
picks = mne.pick_types(raw.info, meg=False, eeg=True, eog=True)
baseline = (-0.1, 0.0)
epochs = mne.Epochs(raw, baseline=baseline, events=events,
                                   event_id=event_id, tmin=tmin,
                                   tmax=tmax, picks=picks,
                                   add_eeg_ref=False, preload=True)
from autoreject import LocalAutoRejectCV
ar = LocalAutoRejectCV()
epochs_clean = ar.fit_transform(epochs)
from mne.preprocessing import ICA
n_components = 25
method = 'fastica'
ica = ICA(n_components=n_components,
          method=method).fit(epochs_clean)
eog_inds, scores = ica.find_bads_eog(epochs_clean, ch_name='EOG')
ica.apply(epochs_clean, exclude=eog_inds)
`
Some ideas to improve this, Now i'm running the autoreject. @agramfort

dengemann, 2017-03-01:
Hi Ramon,

have you actually tried to see if autoreject actually removes your EOG
artifacts? It often/typically does not in our experience.

ramonapariciog, 2017-03-01:
Sorry for the bad presentation of the code Dennis. This is:

`mne.io.set_bipolar_reference(raw, ['EEG 126'], ['EEG 127'], ch_name=['EOG'],
                                              ch_info=dict(ch_types='eog'), copy=False) 

picks = mne.pick_types(raw.info, meg=False, eeg=True, eog=True)

baseline = (-0.1, 0.0)

epochs = mne.Epochs(raw, baseline=baseline, 
                                   events=events, event_id=event_id,
                                    tmin=tmin, tmax=tmax, 
                                    picks=picks, add_eeg_ref=False, preload=True)

from autoreject import LocalAutoRejectCV

 ar = LocalAutoRejectCV() 

epochs_clean = ar.fit_transform(epochs) 

from mne.preprocessing import ICA

n_components = 25 

method = 'fastica' 

ica = ICA(n_components=n_components, method=method).fit(epochs_clean)

eog_inds, scores = ica.find_bads_eog(epochs_clean, ch_name='EOG') `

 About your question, actually, this i toke this from the clean_epochs.plot() :

[image]

Maybe i'm not using autoreject properly.
I will try to run the code again to check it.

jasmainak, 2017-03-01:
@ramonapariciog It is possible that there is a bug with picks for ECG. I usually remove the ECG channel before applying `autoreject`. Could you perhaps reproduce this with the sample data?

larsoner, 2019-08-21:
No response, no data to reproduce, closing

closed 2019-08-21 by larsoner