mne-tools/mne-python #4028: Reading EDF/BDF files with n_records = -1

jpazdera, 2017-03-01:
I've noticed in our lab that the headers of our BDF files often list the number of data records in the recording as -1. Currently, this breaks the read_raw_edf function, as it calculates the number of samples in the recording to be -1 times the number of samples per record.

Although we haven't yet determined what is causing our headers to list the number of records as -1, it is actually possible to determine the number of records in the recording without relying on the header. You can do this by determining the number of bytes in the file that are not part of the header (i.e. the number of data bytes), dividing that number by 3 (as there are 3 bytes per sample), dividing again by the number of channels (to get the number of samples per channel), then dividing by the number of samples per record. This gives the number of records in the recording, as it would be listed in the header. Note that this procedure assumes all channels have the same sampling rate.

Here is some code I added into my own copy of MNE's EDF/BDF reader to get around the issue. The custom code comes right after line 381 in the current version of edf.py on Git.

```
    (...)
    fid.read(32 * nchan).decode()  # reserved
    assert fid.tell() == header_nbytes

    # BEGIN CUSTOM CODE
    if n_records == -1:
        fid.seek(0,2)
        num_bytes = fid.tell()
        num_data_bytes = num_bytes - header_nbytes
        total_samples = num_data_bytes / 3 if subtype == '24BIT' else num_data_bytes / 2
        samps_per_chan = total_samples / nchan
        edf_info['n_records'] = n_records = samps_per_chan / n_samps[0]
    # END CUSTOM CODE

physical_ranges = physical_max - physical_min
cals = digital_max - digital_min
(...)
```
If anyone else is having this trouble, it may be helpful to add some sort of handling like this into the EDF/BDF reader.

Eric89GXL, 2017-03-01:
Can you record some short (e.g, 1 or 2 sec) file to reproduce the problem?

Are you running custom software, or is this straight from the standard acquisition system?

I'm fine with adding this sort of change if you're on a stock setup. It would be good to identify why (acquisition version number maybe?) and put that as a comment in the code.

Actually, perhaps we should *always* do this calculation, and in your case use it as the true value, and in other cases, check against it and warn if there is a mismatch.

jpazdera, 2017-03-01:
I'll try to get a quick recording for you tomorrow.

We're using the standard acquisition system (BioSemi's ActiView software). It's possible this is being caused by some sort of user error when our researchers save our data files. Even so, it's likely we're not the only ones with messed up n_record values.

I do think it could be good to do the calculation, warn if there is a mismatch, and use it as the default if the header says -1.

jpazdera, 2017-03-02:
I tried recording a few short files, but all of them ended up with correct headers. Yet virtually all of our actual recordings end up with -1 as the number of records. I'm working with some of our research staff to figure out what's going on at our end with regards to this matter. I imagine we're probably not the only lab experiencing this issue.

jpazdera, 2017-03-02:
UPDATE: I just figured it out. Our researchers have been **pausing** the recording, then exiting the program. The correct method is to **stop** the recording, then exit the program. I suspect the header item for number of records is not written until the recording is fully stopped, so all our recordings are ending up with the number of records never being written.

Regardless, here's a short recording that has a num_records == -1 error for your testing purposes, now that I know how to recreate the problem:

[numrecs-1.bdf.zip](https://github.com/mne-tools/mne-python/files/815201/numrecs-1.bdf.zip)

Eric89GXL, 2017-03-03:
Are you up for making a PR to fix it?

First step would be to a PR to the [mne-tools/mne-testing-data](https://github.com/mne-tools/mne-testing-data) repo to add the files (probably in the EDF directory). Then you can make a PR to `mne-python` to add the functionality, with a test (which will really just be doing a roundtrip read of the file).

jpazdera, 2017-03-03:
Sure, I'll look into this next week.

cbrnr, 2017-03-30:
I have exactly the same issue with some files here in our lab, thanks @jpazdera for pointing out when this happens! Let me know if I can help in this fix.

closed 2017-04-04 by agramfort