mne-tools/mne-python #4032: Mini-symposium at Scipy

teonbrooks, 2017-03-02:
There's a [call](https://neurostars.org/t/call-for-papers-neuroscience-in-python-mini-symposium-at-the-scientific-computing-in-python-conference-2017-scipy-july-10th-16th-ut-austin/250) for a minisymposium at Scipy on neuroscience in Python. I think it would be a good place to highlight the work from the [Supervised Neural Time Series Sprint](https://kingjr.github.io/supervised_time_series/) in NYC. 

The call ends March 27th, which means we have less than three weeks if we wanted to do so. I will be around this summer to go and can take the lead on the submission if we decide to go for it. Thoughts?

agramfort, 2017-03-03:
I plan to go to scipy to give a sklearn tutorial with Andy so I can also
talk about MNE

@teonbrooks <https://github.com/teonbrooks> are you coming too?

teonbrooks, 2017-03-04:
cool cool. I would like to go. I will be right after my Mozilla fellowship ends so I need to find new support to go.

agramfort, 2017-03-06:
FYI I just submitted a proposal of talk on MNE-Python.

teonbrooks, 2017-03-07:
oh ok, cool. 👍

closed 2017-03-31 by agramfort