mne-tools/mne-python #4034: Problem when calculating stc grand average

mmittag, 2017-03-04:
Hi,

A subject causes problems when I wish to grand average the data with the script attached. The following error message occurs:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/mmittag/anaconda2/lib/python2.7/site-packages/spyder/utils/site/sitecustomize.py", line 866, in runfile
    execfile(filename, namespace)
  File "/home/mmittag/anaconda2/lib/python2.7/site-packages/spyder/utils/site/sitecustomize.py", line 94, in execfile
    builtins.execfile(filename, *where)
  File "/disk_array/ilabs/Firecracker/stc/stc_mean_dys_6mo.py", line 20, in <module>
    avg = avg + stc.data / len(subj)
ValueError: operands could not be broadcast together with shapes (8196,1640) (8195,1640) 
>>> 

@Eric89GXL mentioned to open MNE issue about that. He explained that this can happen when a subject has had a vertex removed during forward computation due to the source being too close to (or outside) the BEM surface. 

Thanks for your advise!
[script_gavg_stc.pdf](https://github.com/mne-tools/mne-python/files/818874/script_gavg_stc.pdf)

agramfort, 2017-03-06:
did you morph all the STCs to fsaverage?

if so it should not crash.

Eric89GXL, 2017-03-06:
No, it's with a morphed anatomy template (using `mne.coreg`), so no need to morph.

Although you do bring up a seemingly trivial solution, which is to do something like `stc.morph(subj, subj, vertices_deficient, vertices_complete, smoothing_steps=1)`. It might actually be worth doing a morph from self->self in all cases, just to get a smoother spatial profile for averaging/stats.

Eric89GXL, 2017-03-06:
I'll add this to the DOC issue, if other have ideas feel free to comment anyway

closed 2017-03-06 by Eric89GXL