mne-tools/mne-python #4036: ENH: autolabel for plot_evoked_topo

jjnurminen, 2017-03-07:
In `plot_evoked_topo()` with multiple conditions, it's usually crucial to know which condition is which. The corresponding example has a piece of matplotlib code that insert labels, but it's a bit clumsy (the user needs to compute label coordinates, etc.)

Should the method provide a facility that would automatically label the conditions according to `evoked.comment` (or some other user supplied-text)?

jjnurminen, 2017-03-08:
If you look at the example here:

http://martinos.org/mne/dev/auto_examples/visualization/plot_topo_compare_conditions.html#sphx-glr-auto-examples-visualization-plot-topo-compare-conditions-py

it has some matplotlib code in the end, which inserts labels for the different conditions onto the plot. I was suggesting that plot_evoked_topo() could automatically insert the labels according to `evoked.comment`.

agramfort, 2017-03-08:
got it. +1 for this.

jjnurminen, 2017-03-08:
Ok. I just wrote a wrapper script for our users that takes multiple evoked files from command line, plots everything and labels according to filename and evoked comment. The result looks like this:

[image]

For automatic labeling in `plot_evoked_topo()`, there are a few questions:

-Since there are different layouts, what's the best default for the label positions?
-It looks like `evoked.info['filename']` is no longer inserted by` read_evoked()`, so filenames are not available just by looking at the evoked set. Does anybody know why?
-I guess there should be an `labels=on|off` kwarg. Maybe it could also take a list of alternative labels from the user.

Eric89GXL, 2017-03-08:
> Since there are different layouts, what's the best default for the label positions?

How hard would it be to add all 9 matplotlib positions? Aesthetically I'm not sure what the best option is. I'd vote lower left.

> It looks like evoked.info['filename'] is no longer inserted byread_evoked()

The filename was never supposed to be part of the measurement info, having it there was a bug. If we need it, we can add it as an `Evoked` class property (just as `self.filename = fname` in the constructor).

> I guess there should be an labels=on|off kwarg. Maybe it could also take a list of alternative labels from the user.

Maybe labels = 'off' or any mpl-like position (e.g., "upper left"). -1 from me on alternative labels for now, let's assume YAGNI / people should set `evoked.comment`.

Also, the legend should be given the same background color (probably via `Rectangle`?) that the plots have, otherwise colors can get lost in the legend that are clearly visible in the plots (e.g., white in this example).

Eric89GXL, 2017-03-08:
I meant to vote lower right, not lower left...

jjnurminen, 2017-03-09:
> How hard would it be to add all 9 matplotlib positions? Aesthetically I'm not sure what the best option is. I'd vote lower left.

I'll try and see. Lower right is kind of the 'standard' option for labels (I think).

>The filename was never supposed to be part of the measurement info, having it there was a bug.

Ok. Maybe the user can just insert the filename into `evoked.comment` if they need it.

>Also, the legend should be given the same background color (probably via Rectangle?) that the plots have, otherwise colors can get lost in the legend that are clearly visible in the plots (e.g., white in this example).

Good point.

PR #4042 (2017-03-21): [MRG+2] add legend to topoplot
Here's a proposal that just uses `plt.legend`. Thus no need to reimplement `location`.

Closes #4036
closed 2017-03-21 by Eric89GXL