mne-tools/mne-python #4037: Auto-generating example images at the bottom of API pages

choldgraf, 2017-03-07:
scikit learn has this nice documentation feature where a few example images are generated on the API page for functions. E.g.: http://scikit-learn.org/stable/modules/generated/sklearn.cluster.KMeans.html#sklearn.cluster.KMeans

I assume this is a sphinx-gallery feature, yeah? Anybody know how this works? It'd be cool if MNE did this as well...I can put in some time to get it working if needed :)

Maybe @agramfort or @Eric89GXL know?

Eric89GXL, 2017-03-07:
We have this feature:

http://martinos.org/mne/dev/generated/mne.preprocessing.maxwell_filter.html#examples-using-mne-preprocessing-maxwell-filter

In practice `sphinx-gallery` has simple discovery mechanisms, though. It looks for `import` statements, whereas in our code most things like STCs come as outputs of functions.

So, to get what you *really* want, I think we'd need to add an option to `sphinx-gallery` to e.g. add a comment line to the examples that allow you to manually specify which classes/functions should be linked to the examples.

choldgraf, 2017-03-07:
ah that must be why I didn't find examples of this in MNE before...hmm, would that be worth a PR in sphinx-gallery?

Eric89GXL, 2017-03-08:
You could tell them the limitation and see what they think about the
workaround. But then also people will need to update examples to use it and
keep it in mind in future PRs, which adds some burden. Might be worth it,
I'm a +0 currently.

If we did have it working, do you see a good set of examples and tutorials
to update that would help the documentation a lot?

choldgraf, 2017-03-08:
One of the devs there mentioned that they're not super thrilled about adding more functionality that would include sphinx-gallery specific comments on rendered HTML, which I get. I'll close this and I guess we can hope that the SG functionality that catches function calls etc gets improved over time :)

closed 2017-03-08 by choldgraf