mne-tools/mne-python #4039: BUG: plotting method cannot accept RGB tuples

jjnurminen, 2017-03-08:
In `plot_evoked_topo()` (and possibly elsewhere), the documentation says that it will take any matplotlib colorspec. However, giving it an RGB tuple results in

    ValueError: third arg must be a format string

from matplotlib. I traced this down to `_plot_timeseries_unified()`, which essentially does

    ax.plot(xdata, ydata, colorspec)

However this call signature does not work when colorspec is an RGB tuple. As the matplotlib docs say, RGB tuples can only be specified as kwargs. IMO a simple fix is to change the above to:

    ax.plot(xdata, ydata, color=colorspec)

The problem also happens at `_plot_timeseries()` (but not elsewhere, according to quick grep)

agramfort, 2017-03-08:
PR welcome.

Alex

jjnurminen, 2017-03-08:
#4040

closed 2017-03-08 by Eric89GXL