mne-tools/mne-python #4045: raw.filter() highpass does not remove DC

mmagnuski, 2017-03-09:
I find it surprising, is it intended?
After I apply the filter, let's say:
```
raw.filter(0.5, None)
```
when I plot the `raw` with `remove_dc=False` I can see big offsets between channels. My expectation was that 0.5 highpass filter will remove the DC, but it doesn't.

Eric89GXL, 2017-03-10:
>  My expectation was that 0.5 highpass filter will remove the DC, but it doesn't.

Mathematically this implies that `raw[:][0].mean(-1)` is non-zero. Is that the case?

> when I plot the raw with remove_dc=False I can see big offsets between channels.

Keep in mind what you describe here is different from removing DC of the whole signal. `remove_dc` removes the DC *within each plotted time segment*, which the filter is not guaranteed to do. If it doesn't make this clear in the docstring, we should update it.

If you want to attenuate the low frequencies further, you need a steeper transition band than the default (which will be 0.5 in this case).

Can you check to see if this is all true?

jasmainak, 2017-03-10:
> If you want to attenuate the low frequencies further, you need a steeper transition band than the default (which will be 0.5 in this case).

Indeed, this is what we also observed for the biomag demo. Our transition band may not be steep enough for the highpass filter ...

Eric89GXL, 2017-03-10:
> Our transition band may not be steep enough for the highpass filter ...

Depends on the tradeoffs you want to make. FYI the 0.5 Hz the filter is already going to be ~13 seconds long, and using a tighter transition band will make it longer. I tried to design the filters to follow the recommendations from the most recent discussions on the issue, at least.

mmagnuski, 2017-03-11:
Ok, I get the thing with plot remove_dc removing the within-window mean - I thought that it removes the overall dc - but I didn't read the docs on this carefully so the problem may be in the user not the docs :)
Still,  even with 0.5 Hz transition band I expected to get stronger attenuation of the slowest drifts.
But inspecting filter properties with `plot_filter` reveals that the attenuation is not that great.
[image]

Setting a transition bandwidth of 0.2 for example fixes the issue. 
[image]

I'm just wondering that maybe the default transition bandwidth should be smaller than lowpass freq. I remember I never had such a surprise in eeglab, even using the new filters written by Widmann (I don't remember his default for low transition bandwidth though).

Eric89GXL, 2017-03-11:
I never had such a surprise in eeglab, even using the new filters written
by Widmann (I don't remember his default for low transition bandwidth
though)

Can you investigate their default behavior? You can check docs, and filter
an impulse for example.

mmagnuski, 2017-03-11:
Yes, I'll take a look.

jjnurminen, 2017-03-13:
Just out of curiosity, why use FIR by default?
IIR can much more easily achieve steep transition bands and I don't see much disadvantage in offline filtering, where you can do forward-backward.
Sorry if it's a stupid question, I haven't followed the mne-python filter discussions at all...

jjnurminen, 2017-03-13:
Oh, I found the comprehensive and interesting discussion at http://martinos.org/mne/stable/auto_tutorials/plot_background_filtering.html
Not sure about the arguments of Widmann et al. though - for example, numerical stability of IIR filters is a non-issue in my experience, as long as the filter order is kept reasonable.

Eric89GXL, 2017-03-13:
>  for example, numerical stability of IIR filters is a non-issue in my experience, as long as the filter order is kept reasonable.

Yeah, with second-order sections the fear of filter instability isn't so bad, even for higher-order IIR filters. But one thing to keep in mind is that, in addition to instability issues (i.e., high-order filters blowing up), even IIR filters that don't blow up will propagate errors differently from how FIR filters do. Recursive operations can tend to magnify numerical errors.

Also, in [Parks and Burress](http://martinos.org/mne/dev/auto_tutorials/plot_background_filtering.html#id24), they point out that in practice, an IIR filter is effectively equivalent to some FIR filter, because beyond some number of samples the numerical noise effectively truncates the impulse response. Given this length (whatever it is), you can build a FIR filter with better tradeoffs, and more options / control over those tradeoffs, a defined stop-band, etc.

mmagnuski, 2017-03-17:
The default implementation of `pop_eegfiltnew` - the current default eeglab filter seems to be v. similar.
The main difference seems to be that eeglab's FIR is not applied forward-backward but applied only once and then the signal is shifted by the filters group delay.

in eeglabs filter the transitions bandwidth is defined as:
```matlab
df = min([max([maxDf * 0.25, 2]) maxDf]);
```
where `maxDf` is equal to lower edge for highpass (so `0.5` in my case).
So the transition band is the same as in mne.
Filter order is set like this:
```matlab
filtorder = 3.3 / (df / EEG.srate); % Hamming window
filtorder = ceil(filtorder / 2) * 2; % Filter order must be even.
```
so it is 6.6 seconds with 0.5 Hz highpass (if I remember correctly that is the same as in mne too).
Filter frequency respose looks like this:
[image]
additionally, some info that is displayed on screen:
```
pop_eegfiltnew() - performing 1651 point highpass filtering.
pop_eegfiltnew() - transition band width: 0.5 Hz
pop_eegfiltnew() - passband edge(s): 0.5 Hz
pop_eegfiltnew() - cutoff frequency(ies) (-6 dB): 0.25 Hz
pop_eegfiltnew() - filtering the data (zero-phase)
```

The main difference is that I do not get any offsets when I plot the signal after filtering in eeglab.

Eric89GXL, 2017-03-17:
> The main difference seems to be that eeglab's FIR is not applied forward-backward but applied only once and then the signal is shifted by the filters group delay.

Ours is not applied forward-backward by default either. `phase='zero'` does this shift, `phase='zero-double'` does a forward-backward equivalent (though under the hood it convolves the FIR filter with its time reversal, and compensates for the delay of this combined filter).

> so it is 6.6 seconds with 0.5 Hz highpass (if I remember correctly that is the same as in mne too).

I think ours *might* be double that... one of us could be off by a factor of 2.

Is our filter length approximately 3301?

> The main difference is that I do not get any offsets when I plot the signal after filtering in eeglab.

This means there is something different in the design, and thus more to figure out.

What does our `plot_filter` look like?

mmagnuski, 2017-03-17:
> Ours is not applied forward-backward by default either. phase='zero' does this shift, phase='zero-double' does a forward-backward equivalent (though under the hood it convolves the FIR filter with its time reversal, and compensates for the delay of this combined filter).

Oh, I meant phase='zero-double', I thought it is the default, thanks for correcting.

> I think ours might be double that... one of us could be off by a factor of 2.
>
> Is our filter length approximately 3301?

yes, you are right, this is what is printed when I do `raw.plot(0.5, None)`:
```
Setting up high-pass filter at 0.5 Hz
l_trans_bandwidth chosen to be 0.5 Hz
Filter length of 3300 samples (13.200 sec) selected
```

> What does our plot_filter look like?

I've pasted mne's plot filter a few posts above.

Eric89GXL, 2017-03-17:
> I've pasted mne's plot filter a few posts above.

Yes but with log x-axis. Can you do linear x-axis / make it more comparable?

mmagnuski, 2017-03-17:
Ah, sure, here it is:
[image]

mmagnuski, 2017-03-17:
I can export filter coefs from eeglab later today and plot on the same figure if it helps.

Eric89GXL, 2017-03-17:
Sure. From the plot it is clear they are doing something different with the filter design.

We basically delegate directly to `scipy.signal.firwin2` by passing the stop-band (value 0) and pass-band (value 1) edges as given by the user. Because of the windowing this makes the responses a bit smoother than you'd want. I *think* EEGLAB also uses the MATLAB equivalent of firwin2, which means they must adjust those bounds somehow... we'll have to investigate further. Thanks for looking into this @mmagnuski.

Eric89GXL, 2017-03-17:
(I can hopefully help look into it next week unless you beat me to it.)

Eric89GXL, 2017-03-20:
Okay it [looks like the difference](https://gist.github.com/Eric89GXL/248b605b3f400a06540d7b733ed92d8e) is between using [firwin](https://docs.scipy.org/doc/scipy-0.18.1/reference/generated/scipy.signal.firwin.html#scipy.signal.firwin) (what EEGLAB effectively uses) and [firwin2](https://docs.scipy.org/doc/scipy-0.18.1/reference/generated/scipy.signal.firwin2.html#scipy.signal.firwin2) (what we have used forever) to design the filters. It looks like the `firwin`/EEGLAB way does give better suppression for the equivalent window length (EEGLAB is buried beneath the "firwin" curve):

[image]

I hate to do this, but we might want another parameter `design='firwin' | 'firwin2'`, and change the default via deprecation to `firwin`. But maybe we should do this after the upcoming (imminent) release so we can test it out thoroughly.

agramfort, 2017-03-21:
+1 to do this after the imminent release

mmagnuski, 2017-03-21:
+1 too, thanks @Eric89GXL for getting to the bottom of this.

closed 2017-04-05 by jaeilepp

jasmainak, 2017-04-05:
Amazing! @jaeilepp we should update the biomag demo now to reflect this. It was a bit weird to earlier explain that we were not using the defaults for the highpass filter

cbrnr, 2017-08-03:
Just wanted to say thanks for fixing this - with the old `firwin2`, ICA does not work, i.e. it produces strange components. `firwin` fixed this and I now get components almost identical to the ones calculated with EEGLAB. A pity that this won't be the default until 0.16 - could/should we add a warning message for ICA users to make sure to use `firwin`?

Eric89GXL, 2017-08-03:
could/should we add a warning message for ICA users to make sure to use
firwin?

How/where would you want to do this?

cbrnr, 2017-08-03:
> How/where would you want to do this?

Either a short note to the deprecation warning (e.g. "Note that firwin2 produces rather shallow filters that might cause adverse effects in applications where e.g. steep high-pass filters are required, such as in preprocessing signals with a 1 Hz highpass filter for ICA fitting.").

Or put such a statement as a note in the filter docs. Or both. Had I known that MNE was using firwin2 earlier, this would have saved me at least 1 day of work (if not more) that it took me to find out why my ICs look so weird.

Eric89GXL, 2017-08-03:
That sounds reasonable to me. Can you make a PR to add notes where you think it would help?

cbrnr, 2017-08-04:
Done, please see #4461.

PR #4113 (2017-04-05): MRG: Better filtering
Closes #4045.

~~WIP because it needs to:~~

- [x] respect different trans_bandwidths, e.g. for band-pass, band-stop, and notch filters (by changing length in the building function).
- [x] add a bit more narrative (preferably with refs?) to the tutorial
- [x] actually test band-stop, band-pass, and notch behaviors
- [x] force `length='auto'` in `firwin` mode (otherwise the transitions won't work properly)
- [x] update lots of tests to not throw warnings
- [x] update `whats_new.rst`

This also fixes a bug where `ref_meg` channels were not being filtered by default in `raw. [... cut]