mne-tools/mne-python #4049: dipole amplitudes inconsistent and plotting issue

joewalter, 2017-03-10:
According to the docstring of our Dipole object, the dipole amplitude must be given in nAm. However, the Dipole class uses Am. This can be seen e.g. in the save method or the new dipole plotting. This leads to problems e.g. when plotting rap music results with the orthoview method, as the rap music output is in nAm. Which is the right way to go? We either need to adapt some lines in dipole.py and viz._3d or the docstring in dipole.py and our rap music code.

Moreover I tried the new orthoview plotting today. First of all, very nice work!! I came across one issue. If _plot_dipole_mri_orthoview is applied for a list of dipoles the if statement dipole.gof in line 1738 in mne.viz._3d.py fails. The bugfix for this is very easy as we simply need move line 1745-1747 before line 1737 (i.e., first call _concatenate_dipoles(dipole)).

Eric89GXL, 2017-03-11:
Units should be Am, so rap music probably has a bug and so does our
docstring

agramfort, 2017-03-11:
+1

PR welcome

closed 2017-03-15 by agramfort