mne-tools/mne-python #4051: Q?: how to add layout to info

kingjr, 2017-03-10:
I'm trying to plot an home-made EEG dataset (e.g. `evoked = mne.EvokedArray([[0, 0], [0, 0]], info=mne.create_info(2, sfreq=1))`)

I made a layout and I can do `evoked.plot_joint(topomap_arg=dict(layout=layout))`. However the plot doesn't show Jona's spatial colors because `plot_joint()` looks for the layout in the evoked instance.

Is there a way to add a `layout` to an evoked/info?

teonbrooks, 2017-03-10:
do you have a Montage or can you make it into one? that could be added with `.set_montage`

kingjr, 2017-03-10:
I'm confused. I thought a montage was used to do like bipolar construct etc.

I just want to include the layout, i.e. be able to plot the data based on the position of the electrodes.

I tried

```python
montage = mne.channels.Montage(layout.pos[sel], np.array(layout.names), 3, range(len(layout.names)))
raw.set_montage(montage)
mne.io.write_info('eeg_layout-info.fif', raw.info)
info = mne.io.read_info('eeg_layout-info.fif')
```

But this just load my RAM forever

kingjr, 2017-03-10:
(i.e. reading the resulting info leads to an infinite loading...)

kingjr, 2017-03-10:
And on the same topic, the following code fails for non-explicit reasons. I think we may need to revision of the Montage functions.

```python

import mne
from mne.io import read_raw_edf
from mne.datasets import eegbci
from mne.channels import read_layout

raw = read_raw_edf(eegbci.load_data(1, [6])[0], preload=True)
raw.rename_channels(lambda x: x.strip('.'))

# Setup layout
layout = read_layout('EEG1005')
montage = mne.channels.Montage(
    pos=layout.pos[:, :2],
    ch_names=layout.names,
    kind='standard_1005',
    selection=range(len(layout.names)))
raw.set_montage(montage)

# Read write
fname = 'eeg_layout-info.fif'
raw.info = mne.io.write_info(fname, raw.info) # ok
info = mne.io.read_info(fname)  # fails

```

Eric89GXL, 2017-03-11:
Yes, see #3893

Eric89GXL, 2017-03-11:
(This is one potential target for me during the sprint.)

larsoner, 2018-04-04:
Closing as a doc problem covered by #3893

closed 2018-04-04 by larsoner