mne-tools/mne-python #4053: Inappropriate warning about BEM resolution

christianbrodbeck, 2017-03-11:
Some BEM functions check the resolution of the BEM model and issue a [warning](https://github.com/mne-tools/mne-python/blob/master/mne/bem.py#L1879) if it is higher than ICO-4.

```
RuntimeWarning: The bem surface has 10242 data points. 5120 (ico grade=4) should be enough 
```

However, the BEM model shipped with MNE for FSAverage has a higher resolution, so it seems inappropriate to raise this warning which suggests that the user should have done something differently. I could look for a way to check the subject name before issuing the warning, or is there a more elegant way?

agramfort, 2017-03-12:
I am confused. I don't think I have a BEM file for fsaverage.

christianbrodbeck, 2017-03-12:
Here: https://github.com/mne-tools/mne-python/tree/master/mne/data/fsaverage

Eric89GXL, 2017-03-12:
Maybe we should downsample it. It should be possible.

@mshamalainen any reason to use an ICO5 BEM for fsaverage instead of the standard ICO4?

agramfort, 2017-03-13:
arrffff. I would then remove the warning especially if you use a single
layer BEM as it's super fast.

christianbrodbeck, 2017-03-18:
Probably the important part is the second part of the message, only added if there are exactly 3 surfaces: "Dense 3-layer bems may not save properly"

Can there ever be a number of surfaces that is not 1 or 3?

agramfort, 2017-03-18:
it can only be 1 or 3 with our code.

christianbrodbeck, 2017-03-18:
So then is the change in #4080 what you suggested?

closed 2017-03-19 by agramfort