mne-tools/mne-python #4054: interactive ssp example broken

jasmainak, 2017-03-12:
I get this error on the latest master:

```py
Traceback (most recent call last):
  File "/home/mainak/anaconda2/lib/python2.7/site-packages/matplotlib/backends/t
    guiEvent=event)
  File "/home/mainak/anaconda2/lib/python2.7/site-packages/matplotlib/backend_bt
    self.callbacks.process(s, mouseevent)
  File "/home/mainak/anaconda2/lib/python2.7/site-packages/matplotlib/cbook.py"s
    proxy(*args, **kwargs)
  File "/home/mainak/anaconda2/lib/python2.7/site-packages/matplotlib/cbook.py"_
    return mtd(*args, **kwargs)
  File "/home/mainak/anaconda2/lib/python2.7/site-packages/matplotlib/widgets.pd
    self.set_active(i)
  File "/home/mainak/anaconda2/lib/python2.7/site-packages/matplotlib/widgets.pe
    func(self.labels[index].get_text())
  File "mne/viz/utils.py", line 257, in _toggle_proj
    params['plot_update_proj_callback'](params, bools)
  File "mne/viz/topomap.py", line 103, in _plot_update_evoked_topomap
    params['time_idx'])] * params['scale']
IndexError: index 104 is out of bounds for axis 0 with size 102
```
for this example: http://www.martinos.org/mne/stable/auto_tutorials/plot_artifacts_correction_ssp.html
cc @jaeilepp

Eric89GXL, 2017-03-12:
Did you click on something to get this? I haven't seen circle complain
about a failure.

jasmainak, 2017-03-12:
Yep, I clicked on the checkboxes to apply the SSP vectors ...

closed 2017-03-16 by agramfort