mne-tools/mne-python #4055: Doc of mne.decoding.Scaler.fit_transform() is not correct

Akshay0724, 2017-03-12:
`fit_transform()` accepts 3D epoch data, but when i run `help(fit_transform)` command in python then it gives-
````
scaler.fit_transform(X, y=None, **fit_params)
Fit to data, then transform it.
Fits transformer to X and y with optional parameters fit_params
and returns a transformed version of X.

Parameters
----------
X : numpy array of shape [n_samples, n_features]
    Training set.

y : numpy array of shape [n_samples]
    Target values.

Returns
-------
X_new : numpy array of shape [n_samples, n_features_new]
    Transformed array.
````
This comes from `fit_transform()` in `TransformerMixin` class.

agramfort, 2017-03-12:
does https://github.com/mne-tools/mne-python/pull/4025
fix your pb?

Akshay0724, 2017-03-13:
I'm still getting the same output. Reason for this I think is that Scaler
is imports TransformerMixin and in this class fit_transform contain this
doc. We should redefine that function in Scaler class and update the doc.

Thanks

agramfort, 2017-03-13:
can you open a PR?

Akshay0724, 2017-03-13:
OK, I can do it.

PR #4061 (2017-03-26): [MRG] Doc of mne.decoding.Scaler.fit_transform() is not correct
Fixes #4055 

**Changes Made:**
I have added `fit_transform()` in `Scaler` class and corrected the doc.
closed 2017-03-26 by agramfort