mne-tools/mne-python #4059: reference channel when importing .bdf data ?

claire-braboszcz, 2017-03-13:
Hi, 

I am wondering whether it would be appropriate to add a warning message or even the option of defining a reference channel when importing EEG Biosemi data with mne.io.read_raw_edf  ?
I am coming from EEGlab and when importing Biosemi data  it indicates that without a reference noise will be added to the data. (see for example this explanation here, taken from [this SCCN page:](https://sccn.ucsd.edu/wiki/Makoto%27s_preprocessing_pipeline) 

> If you data are recorded with Biosemi, you MUST re-reference the data to *ANY* channel (including average channels; by the way, by including the original reference channel(s) you can re-reference to other channel(s) repeatedly without loosing information). This is because Biosemi lets users take care of common-mode rejection offline. What is common-mode noise? You can see non-stationary 60Hz amplitude wax and wane in a remarkably non-stationary way, which could not be modeled and subtracted by CleanLine at all. However, re-referencing erased it perfectly. So I guess that's an example of common-mode noise. 

Most people will probably re-reference their data at some point, but it may be a useful reminder to add to the function. 

Best, 
Claire

agramfort, 2017-03-13:
you mean the file contains a channel used as ref and we don't use it?

claire-braboszcz, 2017-03-13:
No, just that it is especially important for Biosemi data to be re-referenced to any channel not to get extra-noise in the data. Just adding a message reminding people to re-reference after importation may be good.

Eric89GXL, 2017-04-18:
@claire-braboszcz do you have time to add a snippet to the `read_raw_edf` and `RawEDF` doc `Notes` section?

claire-braboszcz, 2017-04-19:
yes, I will do it asap.

agramfort, 2017-07-22:
@claire-braboszcz do you still plan to do this?

JoseAlanis, 2019-01-29:
Hello, I'd be glad to take care of this one if no one has started already.
Just for clarification, I was thinking of something like this:

_Biosemi amplifiers do not perform "common mode noise rejection" automatically. The signals in the EEG file are the voltages between each electrode and CMS active electrode, which still contain some CM noise (50 Hz, ADC reference noise, etc.). Thus, it is advisable to use any single channel (e.g. Cz) on import of BioSemi data to avoid losing signal information. The data can be re-reference (e.g. common average, average of mastoids) after cleaning the data._

This is just roughly braking down the info provided on Biosemis' [FAQs](https://www.biosemi.com/faq/cms&drl.htm)

agramfort, 2019-01-29:
would you put this info here
http://martinos.org/mne/stable/manual/io.html#biosemi-data-format-bdf ?

>

closed 2019-01-31 by cbrnr