mne-tools/mne-python #4070: Read .BVEF files

jbschiratti, 2017-03-14:
(Hello,)

The function `mne.channels.read_montage` currently does not support files with the .BVEF extension (BrainVision Electrodes File). These BVEF files have are XML files which contain the names of the electrodes and their position (in spherical coordinates). [AP-64.txt](https://github.com/mne-tools/mne-python/files/840974/AP-64.txt) is a BVEF file that I converted to TXT so that you can see what it contains. 

Brainstorm has a function which reads such BVEF files (https://github.com/brainstorm-tools/brainstorm3/blob/master/toolbox/io/in_channel_brainvision.m). By looking at their code, I was able to modify the `mne.channels.read_montage` function so that it reads BVEF files. 

If you want, I can make a PR.

Eric89GXL, 2017-03-14:
> By looking at their code, I was able to modify the mne.channels.read_montage function so that it reads BVEF files.

Brainstorm is GPL licensed, so we can't look at their code to make an implementation.

You can email them to ask permission to re-license as BSD. Otherwise we need someone to start from scratch, or find code with a compatible license to work from.

agramfort, 2017-03-14:
reading is trivial XML parsing. No need to ask it's doable without copying
their code

PR welcome

jaeilepp, 2017-04-11:
This got done in https://github.com/mne-tools/mne-python/pull/4109.

closed 2017-04-11 by jaeilepp