mne-tools/mne-python #4072: Fresh download seg faults on unittests

smgutstein, 2017-03-14:
I just cloned a fresh version of mne-python and tried to run the unit tests using `make`.  However, I got a Segmentation fault core dump. I believe I've found the region of the test code that generates the seg fault, but don't really know how to handle it.

Here is what I got the seg fault:

    me@Kansas~/Downloads/temp/mne-python$ make
    rm -rf _build
    find . -name "*.pyc" | xargs rm -f
    find . -name "*.so" | xargs rm -f
    find . -name "*.pyd" | xargs rm -f
    rm -f tags
    find . -name "__pycache__" | xargs rm -rf
    python setup.py build_ext -i
    running build_ext
    rm -f .coverage
    nosetests -a '!ultra_slow_test' mne
    /usr/local/lib/python2.7/dist-packages/scipy/special/__init__.py:627: RuntimeWarning: numpy.dtype    size changed, may indicate binary incompatibility. Expected 96, got 88
      from ._ufuncs import *
      ....
    .[Parallel(n_jobs=2)]: Done  49 out of  49 | elapsed:    0.0s finished
    .[Parallel(n_jobs=2)]: Done  49 out of  49 | elapsed:    0.0s finished
    .[Parallel(n_jobs=2)]: Done  49 out of  49 | elapsed:    0.0s finished
    .[Parallel(n_jobs=2)]: Done 294 out of 294 | elapsed:    0.1s finished
    ..[Parallel(n_jobs=2)]: Done 294 out of 294 | elapsed:    0.0s finished
    .........
       Normal return from subroutine COBYLA
       
       NFVALS =  154   F = 1.313356E-03    MAXCV = 0.000000E+00
       X =-2.792911E-03   1.040626E-02   5.626768E-02   9.024506E-02
      
       Normal return from subroutine COBYLA
      
        NFVALS =  154   F = 1.313356E-03    MAXCV = 0.000000E+00
        X =-2.792911E-03   1.040626E-02   5.626768E-02   9.024506E-02
       
        Normal return from subroutine COBYLA
        
         NFVALS =  181   F = 1.313674E-03    MAXCV = 0.000000E+00
         X =-2.797728E-03   1.042625E-02   5.629723E-02   9.024404E-02
         ....................../usr/local/lib/python2.7/dist-packages/traitsui/ui_traits.py:327: DeprecationWarning:    use "HasTraits.trait_set" instead
        self.set( left = left, right = right, top = top, bottom = bottom )
        /usr/local/lib/python2.7/dist-packages/traitsui/ui_traits.py:327: DeprecationWarning: use      "HasTraits.trait_set" instead
        self.set( left = left, right = right, top = top, bottom = bottom )
      ..Makefile:41: recipe for target 'test' failed 
      make: *** [test] Segmentation fault (core dumped)

The version of git I used was:

      me@Kansas:~/Downloads/temp/mne-python$ git lg
      * 8eb11aa - (HEAD -> master, origin/master, origin/HEAD) doc: fixing python 3 bug in decoding example (#4052) (3 days ago) <Chris Holdgraf>

When I ran `make test-verbose`, the crash seemed to occur in test_report:

     Test mne maxfilter. ... ok
     Test mne report. ... Makefile:45: recipe for target 'test-verbose' failed
     make: *** [test-verbose] Segmentation fault (core dumped)

So, I next isolated the crash to `test_report.test_render_report()`:

     me@Kansas:~/Downloads/temp/mne-python/mne/tests$ python
     Python 2.7.12 (default, Nov 19 2016, 06:48:10) 
     [GCC 5.4.0 20160609] on linux2
     Type "help", "copyright", "credits" or "license" for more information.
     >>> import test_report
     >>> test_report.test_render_report()
    ....
    Rendering : /tmp/tmp6tjvKX/temp-ave.fif
    Traceback (most recent call last):
      File "<stdin>", line 1, in <module>
      File "/usr/local/lib/python2.7/dist-packages/numpy/testing/decorators.py", line 147, in skipper_func
        return f(*args, **kwargs)
      File "/usr/local/lib/python2.7/dist-packages/mne/utils.py", line 827, in dec
        return function(*args, **kwargs)
      File "test_report.py", line 78, in test_render_report
         report.parse_folder(data_path=tempdir, on_error='raise')
       File "<string>", line 2, in parse_folder
       File "/usr/local/lib/python2.7/dist-packages/mne/utils.py", line 708, in verbose
        return function(*args, **kwargs)
      File "/usr/local/lib/python2.7/dist-packages/mne/report.py", line 1276, in parse_folder
        for fname in np.array_split(fnames, use_jobs))
      File "/usr/local/lib/python2.7/dist-packages/mne/report.py", line 1276, in <genexpr>
         for fname in np.array_split(fnames, use_jobs))
      File "/usr/local/lib/python2.7/dist-packages/mne/report.py", line 279, in _iterate_files
        html = report._render_evoked(fname, baseline)
      File "/usr/local/lib/python2.7/dist-packages/mne/report.py", line 1607, in _render_evoked
        img = _fig_to_img(ev.plot, **kwargs)
      File "/usr/local/lib/python2.7/dist-packages/mne/report.py", line 84, in _fig_to_img
        dpi=fig.get_dpi())
      File "/usr/local/lib/python2.7/dist-packages/matplotlib-2.0.0b3+1814.g3cb79ed-py2.7-linux-  x86_64.egg/matplotlib/figure.py", line 1666, in savefig
         self.canvas.print_figure(*args, **kwargs)
      File "/usr/local/lib/python2.7/dist-packages/matplotlib-2.0.0b3+1814.g3cb79ed-py2.7-linux-  x86_64.egg/matplotlib/backend_bases.py", line 2227, in print_figure
        **kwargs)
      File "/usr/local/lib/python2.7/dist-packages/matplotlib-2.0.0b3+1814.g3cb79ed-py2.7-linux-x86_64.egg/matplotlib/backends/backend_agg.py", line 517, in print_png
        FigureCanvasAgg.draw(self)
        File "/usr/local/lib/python2.7/dist-packages/matplotlib-2.0.0b3+1814.g3cb79ed-py2.7-linux-   x86_64.egg/matplotlib/backends/backend_agg.py", line 459, in draw
         self.renderer = self.get_renderer(cleared=True)
       File "/usr/local/lib/python2.7/dist-packages/matplotlib-2.0.0b3+1814.g3cb79ed-py2.7-linux-  x86_64.egg/matplotlib/backends/backend_agg.py", line 476, in get_renderer
        self.renderer = RendererAgg(w, h, self.figure.dpi)
        File "/usr/local/lib/python2.7/dist-packages/matplotlib-2.0.0b3+1814.g3cb79ed-py2.7-linux-x86_64.egg/matplotlib/backends/backend_agg.py", line 93, in __init__
         self._renderer = _RendererAgg(int(width), int(height), dpi, debug=False)
    ValueError: Image size of 271801x2500 pixels is too large. It must be less than 2^16 in each direction.

The crash seems to be caused by line 78 of `test_report,py` the code in that region is as follows:

        report = Report(info_fname=raw_fname_new, subjects_dir=subjects_dir)
        with warnings.catch_warnings(record=True) as w:
            warnings.simplefilter('always')
            report.parse_folder(data_path=tempdir, on_error='raise') # -- Here is line 78 --
        assert_true(len(w) >= 1)
        assert_true(repr(report))

Somehow (I think) a graceful exit is needed when an image is too large (and perhaps a smaller test image, or there's another parameter somewhere I need to reset, or there was an error in letting the image get that big)

Eric89GXL, 2017-03-14:
What is your mne.sys_info? I suspect it's a(n old) matplotlib problem.

smgutstein, 2017-03-15:
I think my matplotlib is current:

    >>> mne.sys_info()
    Platform:      Linux-4.4.0-66-generic-x86_64-with-Ubuntu-16.04-xenial
    Python:        2.7.12 (default, Nov 19 2016, 06:48:10)  [GCC 5.4.0 20160609]
    Executable:    /usr/bin/python

    mne:           0.13.1
    numpy:         1.12.0 {lapack=openblas, blas=openblas}
    scipy:         0.17.1
    matplotlib:    2.0.0b3+1814.g3cb79ed

    sklearn:       0.18
    nibabel:       2.1.0
    nitime:        Not found
    mayavi:        4.5.0
    nose:          1.3.7
    pandas:        0.18.1
    pycuda:        2016.1.2
    skcuda:        0.5.2

smgutstein, 2017-03-15:
In `test_report.py` the backend was originally set to 'Agg', I changed it to 'Qt4Agg', which seemed to help somewhat:

    import matplotlib
   matplotlib.use('Qt4Agg')  # for testing don't use X server

However, now when it ran, there were lots of matplotlib plots that popped up and required me to close them in order for the tests to continue (not all of them, but most). Ultimately, the tests crashed with 3 Errors much later. I reran make using test-verbose instead of test.

I'm attaching the results of test-verbose, in case it should make figuring out what's going on any easier.
[make_results.txt](https://github.com/mne-tools/mne-python/files/843424/make_results.txt)

I'm going to keep trying to figure out how to get this fresh install to pass unit tests, Any ideas/thoughts would be helpful.

PS - Just noticed that in mne/viz/tests/test_epochs.py the backend was set to 'Agg'. Tomorrow, I'll try switching it to 'Qt4Agg' to see if that helps, but if there's a quick way to make it possible for me not to need to close so many plots, it would really help.

Eric89GXL, 2017-03-15:
> I'll try switching it to 'Qt4Agg' to see if that helps, but if there's a quick way to make it possible for me not to need to close so many plots, it would really help.

"Agg" is really the backend you should use if you don't want plots popping up. Otherwise the `show=True` in our plotting functions will make the Qt windows pop up.

I suspect this is really some `matplotlib` bug. It shouldn't produce such large images. Can you go to this line:
```
  File "/usr/local/lib/python2.7/dist-packages/mne/report.py", line 84, in _fig_to_img
    dpi=fig.get_dpi())
```
And maybe have it `print(fig.get_dpi())`? I suspect for some reason this number is way too big.

smgutstein, 2017-03-15:
I think you're right, the problem lies somewhere within Matplotlib. I traced the problem a little further. 

Rendering temp_raw.fif and temp_raw-eve.fif works fine, the problem comes with rendering temp-ave.fif.

As of line 84 in `/usr/local/lib/python2.7/dist-packages/mne/report.py`, dpi is 100. 

The problem seems to come when rendering a blank TextBox (i.e. `Text(0,0,u'blank')`). When `savefig` attempts to print the figure - `self.canvas.print_figure(*args, **kwargs)` - line 1666 of `/usr/local/lib/python2.7/dist-packages/matplotlib-2.0.0b3+1814.g3cb79ed-py2.7-linux-x86_64.egg/matplotlib/figure.py`, it goes to `print_figure` in `backend_bases.py`.

In line 2184 it starts to loop through the artists to determine their bounding boxes:

```
                for a in bbox_artists:
                    bbox = a.get_window_extent(renderer)

```
When it comes to the 4th artist, `Text(0,0,u'blank')`, it calls the get_window_extent method for a Text object, which by line 967 in `/usr/local/lib/python2.7/dist-packages/matplotlib-2.0.0b3+1814.g3cb79ed-py2.7-linux-x86_64.egg/matplotlib/text.py`,  tries to transform the reference point (0,0), to where it should be rendered:

       x, y = self.get_transform().transform_point((x, y))

This transform (in  /usr/local/lib/python2.7/dist-packages/matplotlib-2.0.0b3+1814.g3cb79ed-py2.7-linux-x86_64.egg/matplotlib/transforms.py) occurs in line 1329 using an affine transform matrix of:

```
Affine2D(array([[  2.71785235e+03,   0.00000000e+00,  -2.71158618e+05],
                [  0.00000000e+00,   1.37437118e+17,  -2.01396564e+03],
                [  0.00000000e+00,   0.00000000e+00,   1.00000000e+00]]))
```

which I believe is responsible for the resulting error:

`ValueError: Image size of 271801x2500 pixels is too large. It must be less than 2^16 in each direction.`

I have no idea where that affine transform matrix comes from. I'm afraid my knowledge of matplotlib isn't such that I really know where to even start looking. If you have any suggestions, I'd truly appeciate them.

One random thought I've had is that u'blank' seems like a "special" value for a Text object. Perhaps, that is somehow connected with this odd transform?

Eric89GXL, 2017-03-16:
Can you see if you can reproduce it without using any MNE code? Then we can
more easily file a bug with mpl

smgutstein, 2017-03-16:
I'll try. I'm going to be traveling for the next few days, so won't get to take another crack at this 'til Sun.

smgutstein, 2017-03-19:
I don't know if this helps, but line 75 of `mne-python/mne/tests/test_report.py` is 

    epochs.average().crop(0.1, 0.1).save(evoked_fname)

Apparently earlier versions of this line didn't crop the data. The earlier versions run fine. The version with crop doesn't. If I make the 2nd argument of crop about .102 or higher, `test_report` runs fine, but for a smaller number like .101, it crashes. I suspect there's some sort of an issue with 1 datapoint per channel vs. 2+ datapoints per channel, but don't know.

Eric89GXL, 2017-03-20:
Ahh, I see. Yeah mpl probably does some unprotected division.

Can you open a PR to MNE to fix the test (change crop limit, add comment
that using a single point could trigger an mpl bug)?

It would be best to try to make a minimal example for the mpl folks and
send them a bug report, too. But I'll let your conscience hold you to that
:)

smgutstein, 2017-03-20:
OK - I should get to that tonight or tomorrow.

Eric89GXL, 2017-04-05:
I think this has been fixed by PR

closed 2017-04-05 by Eric89GXL