mne-tools/mne-python #4075: Not valid JSON file?

romquentin, 2017-03-15:
Hi Mne team, 
I am running mne-python script on a cluster at NIH and I got this error:
`RuntimeError: The MNE-Python config file (/home/quentinra/.mne/mne-python.json) is not a valid JSON file and might be corrupted`
Is there any easy fix or advice with that?
Thank you
Romain

jasmainak, 2017-03-15:
can you remove the json file and see if it fixes your problem?

romquentin, 2017-03-16:
It does not fix the problem. the mne-python.json is created again when I launch the script and contains this line:
freesurf_subject_path": "/data/quentinra/WM1/subjects/sub11_ADGGGJAZ_MC
which is the good freesurfer path for the participant...
Thanks!

Eric89GXL, 2017-03-16:
Are you using `mne.set_config`? If not, how is `mne-python.json` being created by the script (it shouldn't really be created manually)? What is the entire contents of the faulty `mne-python.json` file?

romquentin, 2017-03-16:
Yes, I am using 
`from mne.utils import set_config`
`set_config("freesurf_subject_path", freesurf_subject_path)`
Not sure the mne-python.json exists before that, or at least it contains only this line.
I am running several participants on different jobs so maybe this creates conflict in the json file?

Eric89GXL, 2017-03-16:
> I am running several participants on different jobs so maybe this creates conflict in the json file?

Yes probably. The mne config is not meant for this type of use, it's really only for storing MNE-related configuration. You probably should use `os.setenv` or so.

closed 2017-03-16 by Eric89GXL