mne-tools/mne-python #4078: TypeError for keyword arg 'yscale' while using AverageTFR.plot()

piotrdzwiniel, 2017-03-17:
As in the subject. I get `TypeError: plot() got an unexpected keyword argument 'yscale'` while I try to call `AverageTFR.plot()` with additional `yscale='log'` keyword argument or with any other defined argument `linear` and `auto`, as described in the function definition at https://github.com/mne-tools/mne-python/blob/master/mne/time_frequency/tfr.py lines from 982 to 1111.

What is important, `AverageTFR.plot()` is plotted without problems when `yscale` keyword is not used.

Any ideas? I'm sure, that I call `plot()` function for `AverageTFR` object.

jaeilepp, 2017-03-17:
Which matplotlib version are you using? (hint: ``mne.sys_info()``)

piotrdzwiniel, 2017-03-17:
1.5.0

jaeilepp, 2017-03-17:
``yscale`` is a recent addition. Are you on the latest dev-version of mne?

piotrdzwiniel, 2017-03-17:
Nope, I've installed the stable one, i.e. 0.13.1. So it seems, that `yscale` will be available with 0.14.0 or so?

jaeilepp, 2017-03-17:
Yes.

piotrdzwiniel, 2017-03-17:
Got it - thank you very much for the clarification. Thus, I consider issue as solved.

closed 2017-03-17 by piotrdzwiniel