mne-tools/mne-python #4085: ENH: Extract cHPI pos from ctf data

bloyl, 2017-03-22:
CTF data stores the chpi fits (x,y,z) as channels in the data file  HLC0011-* etc.

Extracting a pos structure to be used in other mne calls (motion correction etc) is relatively straight forward.

Before I do PR I wanted to input on where this code should go? 

1) I could add it to the add a public function to chpi.py.  The downside is this would be a public function and would only work with CTF data.
2) I could add it to somewhere in mne.io.ctf

Does anyone have any strong opinions on this?

bloyl, 2017-03-22:
I am also unsure about the coordinates that are in `info['hpi_results']['dig_points']`?

from https://github.com/mne-tools/mne-python/blame/master/mne/io/ctf/info.py#L50-L52 they seem to be in device coords not 'ctf device' coords. so to bring them into head (mne) coords i just apply dev_head_t?

If i want the transform from 'ctf device' coords to head coords is that transfrom somewhere or do i need to recreate it? using this https://github.com/mne-tools/mne-python/blob/master/mne/io/ctf/trans.py#L116-L120 and dev_head_t?

Eric89GXL, 2017-03-23:
> CTF data stores the chpi fits (x,y,z) as channels in the data file HLC0011-* etc.

What do you mean x/y/z, is that x/y/z of each coil? So some transformation is still necessary to get quaternions? It can't be (only) x/y/z of the head unless there is also rotation information stored (unless they ignore that?).

> I could add it to the add a public function to chpi.py.

Yeah something like `ctf_chpi_to_head_pos` (I don't like that name but you get the idea) would be fine. I think we want to stick with the Neuromag convention for head positions (time, pos, quats, etc.) if possible  -- it seems complete enough to rally around.

> they seem to be in device coords not 'ctf device' coords. so to bring them into head (mne) coords i just apply dev_head_t

I haven't wrapped my head around the CTF coordinate stuff. I think you're right, but you should be able to cross-check by comparing the digitization and `info['dev_head_t']` to the first few positions you obtain through whatever conversion you do.

> If i want the transform from 'ctf device' coords to head coords is that transfrom somewhere or do i need to recreate it?

I think relevant information gets extracted by [_make_ctf_coord_trans_set](_make_ctf_coord_trans_set), but it doesn't look like the CTF device transform survives anywhere publicly.

bloyl, 2017-03-23:
So I have the function written. I'm going to do some sanity tests with it later this week, but I'd like to get some other testers with there own data sets. is a PR the best way to do that? I could also share a gist if that is easier ??

;-----------------------------------------------------------------------------------------------------------------------------------
to answer the above questions

> What do you mean x/y/z, is that x/y/z of each coil? 

yeah they store xyz for each coil so you need to compute a transform mapping those locations to the locations stored in the hz file and eventually in `info['hpi_results']['dig_points']`

> I haven't wrapped my head around the CTF coordinate stuff. I think you're right, but you should be able to cross-check by comparing the digitization and info['dev_head_t'] to the first few positions you obtain through whatever conversion you do.

This seems to be handled correctly through the various transforms. To ask a thorny question... Why is there even a special designation for the CTF device coordinate space. When is it useful to relate ctf_device space to another device space (neuromag)? 

This raises another point: what gets put into info['dig'] as cardinal points in the ctf code are not digitized points. they are the HPI locations mapped to head coords. Ctf expects the hpis to be on fiducial landmarks. the digitization (pos file) is done separate from the CTF acquire and is not needed to define the ctf head coords. all of the points that are loaded from the pos file are added as extra points I believe.

this is a case where it would be useful to see some new point types supported in the dig structure for instance the hpi coils both from the MEG fit as well as the digitization. some groups are also doing multiple acquisitions of the hpi and cardinal points with polhemus to look are stability. It would be nice to see that type of information encodable and visualizable somehow.

Eric89GXL, 2017-03-23:
> So I have the function written. I'm going to do some sanity tests with it later this week, but I'd like to get some other testers with there own data sets. is a PR the best way to do that?

If the API is clear then yeah, a PR is a good way to go because then people can easily check it out and interactively comment/iterate.

> yeah they store xyz for each coil so you need to compute a transform mapping those locations to the locations stored in the hz file

Okay so you do need to actually compute the time-varying `dev_head_t`. Makes sense.

> Why is there even a special designation for the CTF device coordinate space. When is it useful to relate ctf_device space to another device space (neuromag)?

IIUC it's because different manufacturers defined their spaces differently, there is no industry gold standard. For us the standard we follow is to use Neuromag device coords, but data can still come in from other device coordinate definitians / systems (and does with CTF data I think), and in those cases we need those transforms.

> this is a case where it would be useful to see some new point types supported in the dig structure for instance the hpi coils both from the MEG fit as well as the digitization. some groups are also [...]

You can in `info['dig']` define the same point in multiple spaces.

For the re-digitization, we could already add redundant points, i.e., those with the same `kind` and `ident`, and let the list ordering define which belongs to each redone set. We'd need to make some downstream functions smarter about picking points. But we should wait until people have a compelling use case for using this sort of thing in production analyses before we foist a heavier API on the MNE ecosystem.

larsoner, 2019-08-20:
I think this is fixed in `master`

closed 2019-08-20 by larsoner