mne-tools/mne-python #4087: set_eeg_reference() using all electrodes, including dropped ones

janfreyberg, 2017-03-23:
This may be the desired behaviour, but I was suprised that when I do the following:
```
epoch.pick_channels(my_channels)
epoch.set_eeg_reference()
```
The reference used is the average reference of all electrodes, including the ones excluded in the pick_channels method.

Is this supposed as intended?

Eric89GXL, 2017-03-23:
No, that shouldn't happen. Can you make a complete copy-paste minimal example using `EpochsArray` or the sample data?

janfreyberg, 2017-03-23:
Yep, sure:

```python
import mne
import numpy as np

info = mne.create_info(10, 256, ch_types='eeg', montage=None)
data = np.random.rand(20, 10, 256*5)
epochs = mne.EpochsArray(data, info)
epochs2 = epochs.copy()
epochs.load_data()
epochs2.load_data()
epochs2.pick_channels([str(n) for n in range(5)])  # pick half the channels

epochs.set_eeg_reference()
epochs2.set_eeg_reference()

if np.all(epochs.get_data()[0, 0, :] == epochs2.get_data()[0, 0, :]):
    print('Rereferenced to dropped electrodes')
else:
    print('Only rereferenced to picked electrodes')
```

This is on: 
```
$ pip freeze | grep mne
mne==0.14.dev0
```

jaeilepp, 2017-03-24:
The eeg reference is not applied yet in your example. You need to call ``epochs.apply_proj`` after setting the reference. Perhaps this should be made more clear in the docs.

janfreyberg, 2017-03-24:
Ah yeah, that's good to know.

However, I still think there's something happening here - the first time I noticed this was during time-frequency analysis so I've written an example using `time_frequency.psd_multitaper`. I assume when you call a time-frequency analysis, all projections are applied?

So this example still produces the behaviour I described - the power spectrum for the first electrode in the full epochs data is the same as the power spectrum as the first electrode in the epochs data with half of all electrodes dropped:
```python
import mne
import numpy as np

info = mne.create_info(10, 256, ch_types='eeg', montage=None)
data = np.random.rand(20, 10, 256*5)
epochs = mne.EpochsArray(data, info)
epochs2 = epochs.copy()
epochs.load_data()
epochs2.load_data()
epochs2.pick_channels([str(n) for n in range(5)])  # pick half the channels

epochs.set_eeg_reference()
epochs2.set_eeg_reference()

psd, _ = mne.time_frequency.psd_multitaper(epochs)
psd2, _ = mne.time_frequency.psd_multitaper(epochs2)

if np.all(psd[0, 0, :] == psd2[0, 0, :]):
    print('Rereferenced to dropped electrodes')
else:
    print('Only rereferenced to picked electrodes')

```

Whereas this example works - when you explicitly say which electrodes to reference to:
```python
import mne
import numpy as np

info = mne.create_info(10, 256, ch_types='eeg', montage=None)
data = np.random.rand(20, 10, 256*5)
epochs = mne.EpochsArray(data, info)
epochs2 = epochs.copy()
epochs.load_data()
epochs2.load_data()
epochs2.pick_channels([str(n) for n in range(5)])  # pick half the channels

epochs.set_eeg_reference()
epochs2.set_eeg_reference([str(n) for n in range(5)])

psd, _ = mne.time_frequency.psd_multitaper(epochs)
psd2, _ = mne.time_frequency.psd_multitaper(epochs2)

if np.all(psd[0, 0, :] == psd2[0, 0, :]):
    print('Rereferenced to dropped electrodes')
else:
    print('Only rereferenced to picked electrodes')

```

janfreyberg, 2017-03-24:
Ah sorry, if I apply the projections it actually works in both cases.

So I think the mistake came from the fact that the reference is applied to the data immediately when provided with a list of reference electrodes, but not when just adding the average - so that's just a confusion on my part.

agramfort, 2017-03-24:
the doc should be I think clarified.

closed 2017-03-24 by Eric89GXL