mne-tools/mne-python #4092: How does the conf limit argument to compare evokeds work?

kambysese, 2017-03-24:
This is likely unrelated to #4091 but still involves behavior of `mne.viz.plot_compare_evokeds` and confidence interval kwarg; here I am still using code from tutorial [gist](http://martinos.org/mne/stable/auto_tutorials/plot_visualize_evoked.html#sphx-glr-auto-tutorials-plot-visualize-evoked-py).

`mne.viz.plot_compare_evokeds(evoked_dict, ci=None, colors=colors, linestyles=linestyles)` returns with trace that contradicts doc string wrt to `ci` kwarg as `float | None`

> Traceback (most recent call last):
>   File "/Users/ktavabi/anaconda2/lib/python2.7/site-packages/IPython/core/interactiveshell.py", line 2881, in run_code
>     exec(code_obj, self.user_global_ns, self.user_ns)
>   File "<ipython-input-4-02b5fa7825f7>", line 2, in <module>
>     colors=colors, linestyles=linestyles)
>   File "/Users/ktavabi/Projects/mne-python/mne/viz/evoked.py", line 1503, in plot_compare_evokeds
>     invert_y=invert_y, axes=ax_, title=title_, show=show))
>   File "/Users/ktavabi/Projects/mne-python/mne/viz/evoked.py", line 1529, in plot_compare_evokeds
>     raise TypeError(msg.format(type(ci)))
> TypeError: "ci" must be float, got <type 'NoneType'> instead.
> 
 
And without a `picks` argument `ci=.09` returns channel type GFP plots without confidence limits. I understand in this case confidence limits don't mean much since this is single subj data, but the input data is a `dictionary` of evoked objects and  the doc string implies a ci limit is calculated across "individual time series". How would I need to setup the evoked `dict` or `list` for data from multiple subjects to get desired ci?

[image]

mmagnuski, 2017-03-25:
@kambysese 
ci is drawn only when the input is *"not None and evokeds is a [list/dict] of lists"*. So if you have a dictionary of evokeds the ci will not be drawn. Only dictionary of lists of evokeds or list of lists of evokeds will lead to ci being drawn.

jona-sassenhagen, 2017-03-28:
Is the documentation unclear @kambysese ? Should we change it?

kambysese, 2017-03-28:
No I stand corrected, documentation makes sense. Thank you. Closing.

closed 2017-03-28 by kambysese