mne-tools/mne-python #4093: How to create custom MEG layouts for channel sub-selections?

kambysese, 2017-03-24:
I am trying to create custom `Vectorview-all` layouts to illustrate channel sub-selection in a paper. For a subset of MEG channels I can get the desired evoked topo plot:

```
import os.path as op
import numpy as np
import mne
from mne.channels.layout import generate_2d_layout

data_path = mne.datasets.sample.data_path()
fname = op.join(data_path, 'MEG', 'sample', 'sample_audvis-ave.fif')
evoked = mne.read_evokeds(fname, baseline=(None, 0), proj=True)
ev = evoked[0].copy()

mask = ~np.in1d(np.asarray(ev.info['ch_names']), np.asarray(ch_selections))
ma_chs_idx = np.where(mask)[0]
ma_chs_names = np.asarray(ev.info['ch_names'])[ma_chs_idx].tolist()
print len(np.where(mask)[0])   # returns 84 for list of MEG channel names in my ch_selections
ev.copy().pick_channels(ma_chs_names).plot_topo()
```
So far so good...
[image]

Now to convert channel selection to layout...

```
VVALL = mne.channels.read_layout('Vectorview-all')
xy = VVALL.pos[:, :2][ma_chs_idx]
names = np.asarray(VVALL.names)[ma_chs_idx]
names.shape() # returns 84
assert np.where(names == np.asarray(ev.copy().pick_channels(ma_chs_names).info['ch_names']))  # passes
generate_2d_layout(xy, ch_names=names.tolist(), ch_indices=ma_chs_idx.tolist()).plot()
```

I am expecting same layout as above, instead I get 
[image]

I am not sure what I am doing wrong here (I get the same result with my own data as well), maybe I am making wrong assumption with `xy = VVALL.pos[:, :2][ma_chs_idx]`? Any ideas greatly appreciated.

jaeilepp, 2017-03-27:
This happens because the layout in the second case doesn't have the full layout information available to define the dimensions. The layouts rarely have the positions in regular head coordinates. I wonder if the right way would be to change that as @Eric89GXL suggested earlier. I'm afraid most of the 'full' layouts would be a bit 'off' after that, since the purpose of having the positions in native coordinates is to make them look nice (I assume).

The other option to solve this particular problem would be to expose more functions and params to the users. That seems a bit more hackish, since it would be up to the user to convert the positions to head coordinates.

jaeilepp, 2017-03-27:
Or maybe the best option would be to allow plotting only a subset of channels with ``layout.plot``.

agramfort, 2017-03-27:
>
> the best option would be to allow plotting only a subset of channels with
> layout.plot.
>

I agree

closed 2017-03-28 by Eric89GXL