mne-tools/mne-python #4094: Avoid throwing a Key Error for unrecognized channel types

smgutstein, 2017-03-26:
In `mne/io/meas_info` around line 1480, there is a check to make certain the channel type is an allowed/recognized type.  I currently have a data file with channel types of  'active electrode', 'straingage', 'temperature sensor', etc. For me, it is easier to force/change these types to 'misc' (similar to the way 'None's are assumed to be 'misc') and then print out a message to the user letting them know this has happened. This way, I can see the EEG results without needing to create edited copies of my data files. 

Would this be generally useful behavior?

agramfort, 2017-03-26:
you suggest to set to misc automatically unknown ch types and raise a
warning?

are the types your using part of the .fif format standard? I mean use
existing FIFF constants?

smgutstein, 2017-03-27:
Yes, that is what I'm suggesting.

I expect the types I'm using don't use the .fif format, since they are not EEG data. Rather, they represent temperature readings, pulse rates and other physiologic signals.

At the moment, just setting them to `misc' seems to work well. I can begin (and I am truly a beginner with EEG signals) to observe the EEG signals I have, without having the program throw a Key Error. It could be that this is not a good long-term solution for me (or anyone else). It might be better just to remove any non-EEG signals from further consideration by MNE-tools, I don't know. 

But for now, this change means I can use MNE-tools, without needing to modify my original data file,  which leaves me wondering if this behavior would be generally useful.

agramfort, 2017-03-27:
ok why not.

can you just tell me what is the original file format of your data? edf?

smgutstein, 2017-03-27:
It came from a .set file.

Should I go ahead and submit a pull request?

agramfort, 2017-03-27:
ok so it should be fixed in the eeglab reader right?

smgutstein, 2017-03-28:
Actually, the fix I'm using is to replace lines 1480+ in mne/io/meas_info.py:

        if kind not in _kind_dict:
            raise KeyError('kind must be one of %s, not %s'
                           % (list(_kind_dict.keys()), kind))

with

        if kind not in _kind_dict:
            print ('kind must be one of %s, not %s' %
                   (list(_kind_dict.keys()), kind))
            print ('  Changing "%s" to "misc"' % kind)
            ch_types[ci] = 'misc'
            kind = "misc"

Does this seem reasonable/generalizable to you?

agramfort, 2017-03-28:
I think the fix should be done in the reader of the file format that is
"too" permissive.
Touching meas_info.py is too intrusive.

smgutstein, 2017-03-28:
OK

jona-sassenhagen, 2017-04-12:
@smgutstein do you want to make a PR for the eeglab reader? (Otherwise @jasmainak  or I will have to do it.)

smgutstein, 2017-04-13:
@jona-sassenhagen - I'm afraid I made a sloppy/newbie-ish mistake. The fix I've made is now under "Changing unrecognized ch_types to 'misc' #4136". 

I have code that (I think) is ready to be pushed. I just need to supply tests that can be released publicly. The data files I used to test on my own haven't been approved for release by my supervisor, but I expect to have releasable files by this weekend, that I can use to create unit tests with.

If you need something sooner or have tests ready to go, please feel free to do what you need to. This is a good learning experience for me, but I don't want to slow anything down here.

closed 2017-07-22 by agramfort