mne-tools/mne-python #4095: UnicodeEncodeError with mne.io.show_fiff

Teekuningas, 2017-03-27:
mne.io.show_fiff cannot handle non-ascii characters in tags. In this case it was the researcher's name (pos 224) that caused the exception: u'Santtu Seip\xe4j\xe4rvi (samaseip)'

Traceback (most recent call last):
    subj_info = mne.io.show_fiff(self._info.get('filename'))
  File "/home/zairex/anaconda2/envs/meggie-env/lib/python2.7/site-packages/mne/io/open.py", line 187, in show_fiff
    read_limit=read_limit, max_str=max_str)
  File "/home/zairex/anaconda2/envs/meggie-env/lib/python2.7/site-packages/mne/io/open.py", line 264, in _show_tree
    out += _show_tree(fid, branch, indent, level + 1, read_limit, max_str)
  File "/home/zairex/anaconda2/envs/meggie-env/lib/python2.7/site-packages/mne/io/open.py", line 264, in _show_tree
    out += _show_tree(fid, branch, indent, level + 1, read_limit, max_str)
  File "/home/zairex/anaconda2/envs/meggie-env/lib/python2.7/site-packages/mne/io/open.py", line 238, in _show_tree
    postpend = ' = ' + str(tag.data)[:max_str]
UnicodeEncodeError: 'ascii' codec can't encode character u'\xe4' in position 11: ordinal not in range(128)

I guess it is a normal convention to do "any object to str" -cast with str, but it has horrible consequences in Finland. To make concatenated string of objects this kind of approach would make our lifes easier: u''.join([unicode(obj) for obj in objects_of_almost_any_kind]).encode('utf-8'). That is, replace every "str" with "unicode" and when there is a real need for encoding, then use .encode('utf-8'). Just a idea..

-- Erkka

agramfort, 2017-03-27:
can you share a file so we can replicate and fix this?

thx

Teekuningas, 2017-03-27:
Sure!

[https://www.dropbox.com/s/kpu6x10jtds5wnh/showfiff_breaks.fif?dl=0](url)

-- Tk

closed 2017-04-03 by Eric89GXL