mne-tools/mne-python #4108: Bug IO CTF

agramfort, 2017-03-28:
When trying to read ArtifactMEG.ds CTF file from:

ftp://ftp.fieldtriptoolbox.org/pub/fieldtrip/tutorial/ArtifactRemoval.zip

I get an error:

RuntimeError: Channel Q11-1706 does not appear exactly once in data

Problem is that channel Q11-1706 is present in the res4['comp'] but not actually recorded.

Eric89GXL, 2017-03-28:
Do you know what the correct fix is? Just discard the channel info / issue a warning with the list of discarded channels?

agramfort, 2017-03-28:
we should remove the non-available channels from the comps

Akshay0724, 2017-03-31:
hi **@agramfort**,
I read the res4 file using it's reader(`_read_res4()`) and found that both comps and chs are `list` of `dict`.

each element res4['comp'] is of form 
`dict(sensor_name=str, ncoeff=int, coeff_type=int, sensors=list, scanno=int)`

while of res4['chs'] is of form
`dict(num_coils,.., ch_name=str,....., coil_type=int, coil=dict)`

what I found that some element of res4['comp'] is containing 'Q11-1706' channel in sensors list.

I think proper fix is that we should replace 'Q11-1706' with empty string.
in other words we should replace the non-available channels in sensors list in comps element with empty string.
Is this is correct fix?

agramfort, 2017-03-31:
you should remove the missing channels from the comp

Akshay0724, 2017-03-31:
hi **@agramfort**,
Sorry to ask but can please tell me what does each element of comps i.e a dict represent.

agramfort, 2017-03-31:
compensation allows to clean channels using external reference channels.
it's a linear operation

closed 2017-05-16 by agramfort